BlueSCSI/BlueSCSI-v2 #4: DS25 and Powerbook Files are 2022_11b

MrTechGadget, 2023-01-26:
The Desktop files and PNG were updated to the 2022.12a revision and blue PNG file, but the DS25 and Powerbook versions are still green and showing the 2022_11b, I'm guessing these are not the latest?

Also, the Desktop folder has a zip file of the gerber files for it, the Powerbook and DS25 versions do not have a zip file of the gerber files.

Thanks, what an exciting day!

androda, 2023-01-26:
Hardware version numbers are based on when the design was created or updated, and will not always be the same across all designs.  For example if I were to create a Centronics 50 Pin design today it would be version "2023_01a", but that would not mean the Desktop, DB25, and PowerBook versions are "out of date".

It does look like we posted some older hardware instead of the latest, and I have pull requests open to resolve that issue.

erichelgeson, 2023-01-26:
Fixed in #5 & #6 - thanks!

closed 2023-01-26 by erichelgeson