BlueSCSI/BlueSCSI-v2 #10: Wrong and misleading use of Open Hardware on bluescsi.com

Juul, 2023-01-27:
On bluescsi.com this hardware is referred to as "open hardware" in several places, however the hardware license you are using is not considered an open hardware license because of the non-commercial clause. Please see the OSHWA definition here: https://www.oshwa.org/definition/ specifically criteria number five.

This requirement that commercial use is allowed is not specific to hardware. In fact the Open Source Definition and Free Software Definition both agree on this: https://opensource.org/osd and https://www.gnu.org/philosophy/free-sw.html#fs-definition

See also "Why isn’t a license that restricts commercial use considered a free culture license?" at the Creative Commons website: https://creativecommons.org/share-your-work/public-domain/freeworks/

and the definition of Free Cultural Works: https://freedomdefined.org/Definition

erichelgeson, 2023-01-27:
Hi @Juul, I understand your criticism, though we are not using OSHWA's license. I had honestly not heard of this organization till a few months ago. I agree the hardware is not "libre" or "free culture license" but it is "open". There are many licenses that grant different rights to the user or the creator. I honestly think there is no good hardware licenses currently for small makers who wish to protect their implementations while being transparent and "open" with their design and hardware.

Juul, 2023-01-28:
Ok I do agree about the lack of good open hardware licenses but the word "open" is absolutely interpreted by the vast majority of developers to mean the same as libre when it comes to licensing. Historically it comes from the term Open Source which was an alternate term promoted by folks who didn't like the term Free Software. Open Source Software or Open Software has always meant software licensed using a license that meets the Open Source Definition and the expectations around what openness means carry over to hardware. 

You will never see the Creative Commons organization refer to the CC BY-NC license as an "open" license. You can ask them directly here if they would advise for or against using the term "open hardware" for something licensed using CC BY-NC: https://creativecommons.org/about/contact/

MrTechGadget, 2023-01-29:
Sorry @Juul I have to disagree that open is "absolutely" interpreted as libre by the vast majority of developers. The term Free is much more likely to be associated with libre. Of course, libre is libre, but open does not mean libre. The CC-BY-NC license is open, but not free of restriction. You are welcome to use it for noncommercial purposes. You can make derivative works and share those with attribution.

According to [Opensource.com](https://opensource.com/resources/what-open-hardware), Open Hardware "refers to the design specifications of a physical object which are licensed in such a way that said object can be studied, modified, created, and distributed by anyone." [CERN's Open Hardware license](https://ohwr.org/cernohl) also "shares the same principles as free software or open-source software: anyone should be able to see the source – the design documentation in the case of hardware – study it, modify it and share it." BlueSCSI under the CC-BY-NC license meets this definition. Does it meet Bruce Perens' definition? No, it does not. But he does not speak for the entire Open Hardware movement.

erichelgeson, 2023-01-29:
I appreciate the debate and understand where you are coming from. We are not "Open Hardware"(tm) but "open hardware". 

From the CC page you linked:
> Why does CC support “non-free” licenses at all? Isn’t that against its mission?
> CC hopes to promote a more open culture than “all rights reserved”. Some creators want to allow very broad use of their works. Some wish to allow some uses but restrict others. We believe that **all levels of openness are worth encouraging** and supporting as an alternative to “all rights reserved.” While we hope that some creators will use the non-free licenses as a stepping stone to greater openness in the future, CC encourages sharing under any of its licenses as a way to create a more open culture.

Emphasis mine - open has many levels. I believe leaving the terms is appropriate and we always emphasize the actual license (be it GPL for the firmware or CC-NC for the hardware).

closed 2023-01-29 by erichelgeson

Juul, 2023-03-01:
From the official wiki page explaining the NC license from the people who wrote it:

_NC licenses do not qualify as “open licenses” under the [Open Definition](http://opendefinition.org/), and works licensed under an NC license are not considered [Free Cultural Works](http://freedomdefined.org/Definition). This may be important if you want others to further distribute your work on Wikipedia, Wikimedia Commons, or other platforms requiring a license that meets the Open Definition or the Definition of Free Cultural Works._