BlueSCSI/BlueSCSI-v2 #13: Expose drives as USB mass storage devices when plugged into host

erichelgeson, 2023-02-04:
Similar to how DietSCSI (BlueSCSI v1 fork) is doing [USB pass through](https://github.com/ArrestedLightning/dietSCSI_firmware/blob/rebrand/src/BlueSCSI.cpp#L581) we could use TinyUSB to present the drive images as usb mass storage devices. Since HFS Fuse has been updated and things like CiderPress/etc can open volumes it would be handy to just be able to plug in your BlueSCSI to your modern or slightly less retro computer and have the HFS/etc volumes mounted.

I've done some work on this and will push a branch soon.

MrTechGadget, 2023-02-04:
This would be awesome and eliminate the whole emulator shuffle workflow I'm fighting right now for my devices that don't have ethernet to access my NAS.

Grieverheart, 2023-10-25:
Would it be possible to pass through other SCSI devices, like a zip drive?

crackmonkey, 2023-10-26:
> Would it be possible to pass through other SCSI devices, like a zip drive?

Not really in this future feature. The initiator mode hardware that @androda is working on should be able to, eventually.

ewanuno, 2024-11-19:
i'm, very interested in this. this is a holy grail for supporting vintage samplers (akai, roland , emu, etc)with modern systems. as it stands there are several bits of software used for managing these samplers which are hard to use on modern computers.

As it stands the options are: 
Desktop pc with a PCI scsi card,
Old laptop with a pcmcia cardbus scsi card, 
Virtualised system with a usb scsi interface.  (the only one that people seem to have success with is the Ratoc U2SCX)
Using vintage computers.

pci and cardbus scsi cards are getting harder to find and usb scsi devices are impossibly expensive

fortunately the Ratoc U2SCX interface has fully functional linux drivers.
it has drivers for Windows 98/98SE/MillenniumEdition/2000/XP/XPx64, AND Mac OS X 10.2.7 or later, 10.3.x, 10.4.x

https://www.synchrotech.com/product-usb/usb2-uscsi-conv_01.html
https://lkml.iu.edu/hypermail/linux/kernel/2002.0/00325.html

akuker, 2024-11-20:
I think we're getting two distinct functions blended in this ticket. **THIS** ticket will be for exposing disk IMAGES to the USB host.

[image]

Issue #212 will be for updating BlueSCSI to "bridge" physical SCSI devices to a USB host.

erichelgeson, 2025-06-19:
Will be in the next release: https://github.com/BlueSCSI/BlueSCSI-v2/releases/tag/2025-June-PreRel