BlueSCSI/BlueSCSI-v2 #18: Atari mega STE crashes

Fbeen, 2023-03-21:
Hi same SD card, Same hdd image works in on the old BlueSCSI board but not on the V2 with the pico. The STE crashes after it has loaded the harddisk driver during booting. Got four bombs on the display :-) I have termination off. If i put termination on the systems does not recognize the bluescsi device at all.

Is there anything i can try? Can i do some debugging to provide more information?

Cheers!

Fbeen, 2023-03-22:
> I'm not a Atari expert but I've heard the Mega STE requires these settings in `bluescsi.ini`
> 
> ```ini
> [SCSI]
> MapLunsToIDs=1 ; On
> Parity=0 ; Off
> ```
> 
> You can also generate this file on https://ini.bluescsi.com/
> 
> Let us know if that gets you past the issue. If not I can ask some some Atari users in our Discord to chime in here.

Thank you! After all it seems to be that the hard disk driver from https://atari.8bitchip.info/pphdr.php doesn´t work on the bluescsi 2. I am thinking about purchasing the drivers from https://www.hddriver.net/ but i find them a little expensive.
Is there anybody who have suggestions?

fdanapfel, 2023-03-27:
There are also some free hardisk drivers for Atari available, like AHDI or CBHD. See https://temlib.org/AtariForumWiki/index.php/IDE_%26_SCSI_Drivers for more information.
(disclaimer: since I don't own a BlueSCSI V2 (yet) I can't say how well any of them will work)

Fbeen, 2023-03-27:
Thank you Frank Danapfel, if time is available then i will try if there is one that works.

mynameistroy, 2023-04-30:
I've personally tested the ICD and HDDRIVER drivers on a MegaSTE. The ICD ones are abandonware at this point and readily available on the internet.

z80micro-mc, 2023-05-02:
Take a look at #31 low capacity SD not recognised by V2 but is ok on v1.

Ditto #27 There is a bug in Parity handling on V2, there is a development patched version that works.
The RM Nimbus also needs parity to be disabled. There are 2 issues with v2: a) 'EnableParity=' does not woork - use 'Parity=0'
b) Although it detects and reports that parity is disabled it still enforces aprity checking. There is a patch version that fixes this I suggest that you give it a go.

Peter

thedocbwarren, 2023-08-31:
Hello, I wanted to see if we can get a followup.  I've upgraded to latest (non-beta) firmware for V2 with no luck.  Log now says:

=== Finding images in / ===
== Opening /HD00_512 1024m.img for ID: 0 LUN: 0
---- WARNING: This image does not appear to be a valid Macintosh Device image. See: https://github.com/BlueSCSI/BlueSCSI-v2/wiki/Disk-Images
---- Image ready

Obviously this is not for a Mac.  It was created via DD.  Using the ICD latest drivers is says unrecognizable device for Lun 0.

mynameistroy, 2023-09-01:
The message is just information and doesn't affect function. Are you trying this with a MegaSTE?

I've used it with my MegaSTE and TT without issues.

Termination should be disabled and EnableParity=0 in the ini file or not present

mynameistroy, 2023-09-01:
Also an update I looked into trying ppera's disk driver and now it is no longer free it would seem

thedocbwarren, 2023-09-01:
Yes I am.  Got closer.  I had termination on.  Does this give any info on why I can't format the image.  It's detected with ICD drivers (1994 version.)

It's a 1GB image with 512mb sectors.  I'm using a Mega STe running TOS 2.06

[10ms] Platform: BlueSCSI Pico
[10ms] FW Version: 2023.08.10-release Aug 11 2023 01:25:40
Flash chip size: 2048 kB
 
=== SD Card Info ===
SD card detected, FAT32 volume size: 29648 MB
SD Name: SL32G, MID: 0x03, OID: 0x50 0x54
 
=== Global Config ===
Reading configuration from bluescsi.ini
Active configuration:
-- MapLunsToIDs is on
 
=== Finding images in / ===
== Opening /HD00_512 1024m.img for ID: 0 LUN: 0
---- WARNING: This image does not appear to be a valid Macintosh Device image. See: https://github.com/BlueSCSI/BlueSCSI-v2/wiki/Disk-Images
---- Image ready

=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- ROM drive image not detected
 
=== Configured SCSI Devices ===
* ID: 0, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 1048576kB
 
Initialization complete!
INFO: Pico Voltage: 3.157V.
WARNING: Host used command 0x1A which is affected by drive geometry. Current settings are 63 sectors x 255 heads = 16065 but image size of 2097152 sectors is not divisible. This can cause error messages in diagnostics tools.

mynameistroy, 2023-09-01:
Does ICDFormat give an error or something when you try formatting it?

thedocbwarren, 2023-09-01:
Yes, It fails during the scan and with a error is can't read some sector.   All I can think is it seems there is a major incompatibility with the way the "geometry" is expressed and what the image is setup as or the firmware is capable of.  It definitely can see the drive and even sees the correct size.  The log shows something is up with the geometry.  This solution is a bit black box so not sure how to adjust this.

thedocbwarren, 2023-09-01:
Maybe if I can get an image size that matches and maybe a correct way to build it.  Hoping to solve this and have this issue as an example for anyone wanting to do this.

thedocbwarren, 2023-09-04:
Ok got it.  It was in deed the image size.  It can't handle an image larger than 2000 512b sectors.  Cam about somewhere in the 512/524mb range.  

I formatted a large partition and have been fine with this.

viking272, 2024-01-21:
I spoke through email with Peter P on his Atari disk driver and Blue SCSI. He would take a look to see what is happening. Do you think developer editions of blue scsi v2 exist or should he or a benefactor buy him one??
I did notice however his drivers support IDE and ACSI only, which I'm assuming they should work on the Mega STE but TT has scsi only so the driver may not be compatible?

thedocbwarren, 2024-01-21:
> I spoke through email with Peter P on his Atari disk driver and Blue SCSI. He would take a look to see what is happening. Do you think developer editions of blue scsi v2 exist or should he or a [...]

I used a nightly build and had success.  It’s the image size not getting translated correctly.  Creating an image shy of 512mb will do the trick.  Think I made it 500mb and it worked.

viking272, 2024-01-21:
Interesting, when you say you use a nightly build is this the latest firmware build which you flash on to the blue scsi v2?

erichelgeson, 2024-05-22:
Is there anything else to follow up on for this ticket? Seems to work for others, but if you are still having an issue please feel free to open another discussion or join or discord for any additional help.

closed 2024-05-22 by erichelgeson