BlueSCSI/BlueSCSI-v2 #21: Dataloss at PowerBook board (possible solution provided)

schreinerman, 2023-04-01:
Currently I have data losses at PowerBook 170. 

I will try to have a fix for this issue.

1) I'm preparing to add additional capacitors to 5V (2x 6.3V/1000uF), 3.3V (3x 6.3V/1000uF) and 2.8V (1x 6.3V/1000uF).
2) Raspberry Pi Pico can detect via GPIO24 if a voltage is present, GPIO29 (ADC3) can measure the voltage via ADC. If the VSYS voltage drops below 3V, the SD-Card should be synced and read/write should be stopped to prevent data losses.

androda, 2023-04-01:
Hello,

It sounds like you are planning to add capacitance to the various voltage rails.  Have you done any measurements of the input 5v sources and the output 3.3 / 2.85 rails to see if and how much they may be drooping?

schreinerman, 2023-04-01:
I did not made any measurements yet. Since the Pico can be powered from 1.8 - 5.5V, also lower voltages are not a big issue. The Pico contains a buck-boost converter which is stabilizing the voltage to 3.3V. 2.85V should normally also not the big deal while around 4V is required as input voltage.

Anyhow: With all the capacitors I don't have any data losses anymore.

See some picture attached.

[image]

androda, 2023-04-25:
I'm glad you were able to resolve the issue.  This is the first and only report of data loss on the PowerBook version, and I have been unable to duplicate it.

closed 2023-04-25 by androda