BlueSCSI/BlueSCSI-v2 #22: New Feature: Dynaport SCSI Network Card Emulation

schreinerman, 2023-04-01:
PiSCSI can emulate a Dynaport SCSI Network Card. 

If BlueSCSI is used with Pico W, something similar could also work for BlueSCSI.

erichelgeson, 2023-08-05:
This is now in beta https://github.com/BlueSCSI/BlueSCSI-v2/wiki/WiFi-DaynaPORT

schreinerman, 2023-08-12:
Tested today and what to say: works great with my PowerBook 170!

schreinerman, 2023-08-13:
Maybe some comment: I have the BlueSCSI v2 running in my Powerbook 170 and switching the SD-Card is somehow challenging 😉. It would be great, if there would be the possibility to backup the HDD files via WiFi to a FTP / WebDAV server or otherwise to provide an FTP / WebDAV server on the RP2040. Anyhow that's a very high-level request...

erichelgeson, 2023-10-13:
https://github.com/BlueSCSI/BlueSCSI-v2/releases/tag/v2023.10.12

closed 2023-10-13 by erichelgeson