BlueSCSI/BlueSCSI-v2 #26: How can I test the assembled device?

Piligrim-77, 2023-04-25:
Hi, I have assembled 4 pieces of BlueSCSI V2 devices and only two of them work without problems. 
On the other two devices, there is a problem when trying to initialize the hard drive. 
Is there any program for testing chips on the board, or maybe there is a way to determine which chip is the problem from the logs?
______________________________________________
Log:
"[10ms] Platform: BlueSCSI Pico
[10ms] FW Version: 2023.04.10-release Apr 10 2023 19:08:58
Flash chip size: 2048 kB
 
=== SD Card Info ===
SD card detected, FAT32 volume size: 3832 MB
SD Name: SD16G, MID: 0xFE, OID: 0x34 0x32
 
=== Global Config ===
Reading configuration from bluescsi.ini
Active configuration (using system preset "Generic"):
-- SelectionDelay: 0
-- ReinsertCDOnInquiry is enabled
 
=== Finding images in / ===
== Opening /HD00_512-1GB.hda for ID: 0 LUN: 0
---- Image ready

=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- ROM drive image not detected
 
=== Configured SCSI Devices ===
* ID: 0, BlockSize: 512, Type: Fixed, Quirks: None, Size: 1048576kB
 
Initialization complete!
SCSI ID 0 negotiated synchronous mode 5 MB/s (period 4x50 ns, offset 8 bytes)
scsi_accel_rp2040_stopWrite() timeout, FIFO levels 0 2 PC 17"
______________________________________________

Log with debugging enabled:
"[10ms] Platform: BlueSCSI Pico
[10ms] FW Version: 2023.04.10-release Apr 10 2023 19:08:58
Flash chip size: 2048 kB
 
=== SD Card Info ===
SD card detected, FAT32 volume size: 3832 MB
SD Name: SD16G, MID: 0xFE, OID: 0x34 0x32
[22ms]  
[22ms] === Global Config ===
[23ms] Reading configuration from bluescsi.ini
[24ms] Active configuration (using system preset "Generic"):
[25ms] DBG -- SelectionDelay: 255
[25ms] DBG -- EnableUnitAttention is off
[26ms] DBG -- EnableSCSI2 is on
[27ms] DBG -- EnableSelLatch is off
[27ms] DBG -- MapLunsToIDs is off
[28ms] -- Debug is enabled
[29ms] DBG -- Parity is enabled
[29ms] -- ReinsertCDOnInquiry is enabled
[108ms]  
[108ms] === Finding images in / ===
[108ms] == Opening /HD00_512-1GB.hda for ID: 0 LUN: 0
[386ms] DBG ---- Read prefetch enabled: 8192 bytes
[387ms] ---- Image ready
[392ms] 
[392ms] === ROM Drive ===
[393ms] Platform supports ROM drive up to 1692 kB
[393ms] ---- ROM drive image not detected
[393ms]  
[394ms] === Configured SCSI Devices ===
[394ms] * ID: 0, BlockSize: 512, Type: Fixed, Quirks: None, Size: 1048576kB
[496ms]  
[496ms] Initialization complete!
[16279ms] DBG BUS RESET
[23529ms] DBG ---- SELECTION: 0
[23530ms] DBG ---- MESSAGE_OUT
[23531ms] DBG ------ OUT: 0xC0 
[23532ms] DBG ------ OUT: 0x01 
[23532ms] DBG ------ OUT: 0x03 
[23533ms] DBG ------ OUT: 0x01 
[23533ms] DBG ------ OUT: 0x32 
[23534ms] DBG ------ OUT: 0x08 
[23534ms] SCSI ID 0 negotiated synchronous mode 5 MB/s (period 4x50 ns, offset 8 bytes)
[23535ms] DBG ---- MESSAGE_IN
[23535ms] DBG ------ IN: 0x01 0x03 0x01 0x32 0x08 
[23538ms] DBG ---- COMMAND: Inquiry
[23538ms] DBG ------ OUT: 0x12 0x00 0x00 0x00 0x02 0x00 
[23539ms] DBG ---- DATA_IN, syncOffset 8 syncPeriod 50
[28540ms] scsi_accel_rp2040_stopWrite() timeout, FIFO levels 0 2 PC 17
[28540ms] DBG ---- Total IN: 2 OUT: 0 CHECKSUM: 0
[28541ms] DBG ---- STATUS: 0 GOOD"
______________________________________________

Piligrim-77, 2023-04-25:
I ordered PCB boards without components in JLCPCB and soldered all the parts myself. The details were new from the offline store. I don't quite understand about the "target test system", could you tell me what it is? 
I would like to somehow test the passage of signals through the buffer chips on the BlueSCSI board.
Maybe there is some kind of test firmware for Raspberry Pico that sets the state of all outputs to a high or low state. 
I have an oscilloscope and a logic analyzer.

erichelgeson, 2023-04-25:
Could use an approach like PiSCSI does with a loopback tester https://github.com/PiSCSI/piscsi/wiki/Troubleshooting#user-content-Loopback_Testing

daleksec197, 2025-04-13:
Good to know I'm not the only one experiencing this issue...

I’d consider myself fairly experienced when it comes to soldering SMD components, as it’s part of my job.
I ordered the PCB from JLCPCB and initially had trouble with missing parts, so I decided to source them myself and solder everything at home. Other people read books in the evening ;)

My first mistake was buying the wrong transceivers - I used LVT125 & LVT245 instead of the bus-hold variants. Please consider updating the BOM and the schematics - the current setup is a bit confusing with only a note in the schematics. Maybe @Piligrim-77 made the same mistake?

The first board I assembled jammed the bus: my Mac LC II wasn’t able to detect any other SCSI devices. I swapped the ICs, but something else must have gone wrong, so I figured it would be quicker to assemble a second board - using the right ICs. It worked, and I went to bed happy.

The next day, I was working on the LC for about an hour when it suddenly stopped responding. It froze, and I had to power-cycle the system. Now I’m facing the same issue as with the first board - nothing on the bus is being detected.

My setup: the aforementioned LC II with an internal BlueSCSI v2 2022.12a (bus-powered), and the devices being tested with an adaptor (which works flawlessly on every other SCSI device). I’m using internal/50pin versions, v2 2023.10a, likewise on bus power.

At first, I suspected a power issue, but I’ve tried another internal BlueSCSI (same batch, 2022.12a), which has run for hours without any problems.

I couldn't find the exakt regulator so replaced it with an ADP3338, which seems to work.
I’ve already checked termination and voltages - everything seems fine. I plan to do some proper testing at work with my oscilloscope and logic analyser....

I’m also planning to build the loopback tester at some point and run some tests with it. Unfortunately, I won’t be home much over the next couple of weeks, so I won’t be able to follow up right away. But I’d be happy to help Piligrim-77 test with the loopback setup - maybe it’ll help us get to the bottom of these issues.