BlueSCSI/BlueSCSI-v2 #27: BlueSCSI.ini parity option ignored

z80micro-mc, 2023-04-28:
EnableParity option is totally ignored.

EnableParity = 0

or 

EnableParity = 1

Both have the same effect. Parity checking is permanently enabled.

RM kit is an 8 bit SCSI controller whaich does not support parity.  There Parity checking needs disabling.

WIKI  for V2 is very sparse - no information on BLUESCSI.INI configuration options. Nor details on what the various System types do.

Peter

erichelgeson, 2023-04-28:
Looks like a bad merge renamed it to `Parity` - I'll fix it for next release, but should be a work around for the current release.

You can find some details in the examples bluescsi.ini in the repo or an interactive builder at https://ini.bluescsi.com

> WIKI for V2 is very sparse - no information on BLUESCSI.INI configuration options. Nor details on what the various System types do.

It's a wiki so contributions welcome!

z80micro-mc, 2023-04-28:
Eric,

thanks.

It now claims to have disabled Pareity checking but clearly hasn't.

extract from log
[10ms] Platform: BlueSCSI Pico
[10ms] FW Version: 2023.04.10-release Apr 10 2023 19:08:58
Flash chip size: 2048 kB

=== SD Card Info ===
SD card detected, FAT32 volume size: 14776 MB
SD Name: SD16G, MID: 0x27, OID: 0x50 0x48
[21ms]
[21ms] === Global Config ===
[22ms] Reading configuration from bluescsi.ini
[22ms] Active configuration (using system preset "Generic"):
[23ms] -- SelectionDelay: 250
[24ms] -- EnableUnitAttention is on
[24ms] -- EnableSCSI2 is off
[25ms] DBG -- EnableSelLatch is off
[25ms] DBG -- MapLunsToIDs is off
[26ms] -- Debug is enabled
[26ms] -- Parity is disabled
[27ms] DBG -- ReinsertCDOnInquiry is disabled
[102ms]
[102ms] === Finding images in / ===
[103ms] == Opening /HD10_512.HDA for ID: 1 LUN: 0
[109ms] DBG ---- Read prefetch enabled: 8192 bytes
[110ms] ---- Image ready
[115ms]
[115ms] === ROM Drive ===
[115ms] Platform supports ROM drive up to 1692 kB
[116ms] ---- ROM drive image not detected
[116ms]
[116ms] === Configured SCSI Devices ===
[117ms] * ID: 1, BlockSize: 512, Type: Fixed, Quirks: XEBEC, Size: 
41040kB
[218ms]
[218ms] Initialization complete!
[5395ms] DBG ---- SELECTION: 1
[5646ms] DBG scsi_accel_rp2040_finishRead(0x00 0x00 0x00 0x00 0x00 0x00 
) detected parity error
[5647ms] DBG ---- COMMAND: TestUnitReady
[5647ms] DBG ------ OUT: 0x00 0x00 0x00 0x00 0x00 0x00
[5648ms] DBG ---- STATUS: 0 GOOD
[5653ms] DBG ---- MESSAGE_IN
[5653ms] DBG ------ IN: 0x00
[5654ms] DBG -- BUS_FREE
[5654ms] DBG ---- SELECTION: 1
[5904ms] DBG scsi_accel_rp2040_finishRead(0x00 0x00 0x00 0x00 0x00 0x00 
) detected parity error
[5905ms] DBG ---- COMMAND: TestUnitReady
[5906ms] DBG ------ OUT: 0x00 0x00 0x00 0x00 0x00 0x00
[5906ms] DBG ---- STATUS: 0 GOOD
[5913ms] DBG ---- MESSAGE_IN
[5913ms] DBG ------ IN: 0x00
[5914ms] DBG -- BUS_FREE
[5914ms] DBG ---- SELECTION: 1
[6164ms] DBG scsi_accel_rp2040_finishRead(0x08 0x00 0x00 0x00 0x01 0x00 
) detected parity error
[6165ms] DBG ---- COMMAND: Read6
[6166ms] DBG ------ OUT: 0x08 0x00 0x00 0x00 0x01 0x00
[6166ms] DBG ------ Read 1x512 starting at 0
[6167ms] DBG ---- DATA_IN
[6169ms] DBG ---- Total IN: 512 OUT: 0 CHECKSUM: 64268
[6169ms] DBG ---- STATUS: 0 GOOD
[6176ms] DBG ---- MESSAGE_IN
[6176ms] DBG ------ IN: 0x00
[... 93 lines cut ...]
SelectionDelay = 250   # Millisecond delay after selection, 255 = 
automatic, 0 = no delay
PhyMode = 1   # 0: Best available  1: PIO  2: DMA_TIMER  3: GREENPAK_PIO 
4: GREENPAK_DMA
#Dir = "/"   # Optionally look for image files in subdirectory
#Dir2 = "/images"  # Multiple directories can be specified Dir1...Dir9
DisableStatusLED = 0 # 0: Use status LED, 1: Disable status LED
# Settings that can be needed for compatibility with some hosts
Quirks = 4   # 0: Standard, 1: Apple, 2: OMTI, 4: Xebec, 8: VMS
EnableUnitAttention = 1 # Post UNIT_ATTENTION status on power-on or SD 
card hotplug
EnableSCSI2 = 0 # Enable faster speeds of SCSI2
EnableSelLatch = 0 # For Philips P2000C and other devices that release 
SEL signal before BSY
Parity = 0

MapLunsToIDs = 0 # For Philips P2000C simulate multiple LUNs
MaxSyncSpeed = 0 # Set to 5 or 10 to enable synchronous SCSI mode, 0 to 
disable

Peter

(Issue #27)

<https://ini.bluescsi.com>

z80micro-mc, 2023-04-28:
There are 2 aspects to Parity:

Parity genration
Parity Checking.

These should be on separate flags or Parity shoud have a state setting:  Off, generate but ignore, generate and check

Peter

mynameistroy, 2023-04-30:
I'm not sure if I'm understanding this correctly but does this host actually care if parity is generated? 

Parity is usually referred in the target device domain as ignored or checked that I am familiar with anyways. So it would only need a toggle between 2 states.

When parity is disabled it is actually failing the transfers or is it just reporting them in the debugging log as acknowledging that parity is invalid? ie just noise?

If it's failing the transfers than that is definitely still a problem!

z80micro-mc, 2023-04-30:
Troy,

The BlueSCSI log.txt file ('Debug=1') have an entry at the end of each 
command that says that a parity error has been detected and operations 
fail.

The host neither generates nor checks parity (DP) is nc. This functions 
fine with most drives in particular the Quantum Pro40S has a jumper 
'EP', if the link is present the drive checks parity (enable parity), if 
the link is missing then it ignores parity.

BlueSCSI v2 code seems to still check parity despite 'Parity=0', setting 
to ignore it.

----------
log.txt extract

[10ms] Platform: BlueSCSI Pico
[10ms] FW Version: 2023.04.10-release Apr 10 2023 19:08:58
Flash chip size: 2048 kB

=== SD Card Info ===
SD card detected, FAT32 volume size: 14776 MB
SD Name: SD16G, MID: 0x27, OID: 0x50 0x48
[20ms]
[20ms] === Global Config ===
[21ms] Reading configuration from bluescsi.ini
[22ms] Active configuration (using system preset "Generic"):
[22ms] -- SelectionDelay: 150
[23ms] DBG -- EnableUnitAttention is off
[24ms] -- EnableSCSI2 is off
[24ms] DBG -- EnableSelLatch is off
[25ms] DBG -- MapLunsToIDs is off
[25ms] -- Debug is enabled
[25ms] -- Parity is disabled
[26ms] DBG -- ReinsertCDOnInquiry is disabled
[88ms]
[88ms] === Finding images in / ===
[88ms] == Opening /HD10_512.HDA for ID: 1 LUN: 0
[95ms] DBG ---- Read prefetch enabled: 8192 bytes
[95ms] ---- Image ready

...

[202ms] Initialization complete!
[5462ms] DBG ---- SELECTION: 1
[5613ms] DBG scsi_accel_rp2040_finishRead(0x00 0x00 0x00 0x00 0x00 0x00 
) detected parity error
[5614ms] DBG ---- COMMAND: TestUnitReady
[5614ms] DBG ------ OUT: 0x00 0x00 0x00 0x00 0x00 0x00
[5615ms] DBG ---- STATUS: 0 GOOD
[5621ms] DBG ---- MESSAGE_IN
[5622ms] DBG ------ IN: 0x00
[5623ms] DBG -- BUS_FREE
[5623ms] DBG ---- SELECTION: 1
[5773ms] DBG scsi_accel_rp2040_finishRead(0x00 0x00 0x00 0x00 0x00 0x00 
) detected parity error

...

[5937ms] DBG ---- COMMAND: Read6
[5938ms] DBG ------ OUT: 0x08 0x00 0x00 0x00 0x01 0x00
[5938ms] DBG ------ Read 1x512 starting at 0
[5939ms] DBG ---- DATA_IN
[5941ms] DBG ---- Total IN: 512 OUT: 0 CHECKSUM: 64268
[5941ms] DBG ---- STATUS: 0 GOOD
[5950ms] DBG ---- MESSAGE_IN
[5950ms] DBG ------ IN: 0x00
[5952ms] DBG -- BUS_FREE
[5952ms] DBG ---- SELECTION: 1
[6102ms] DBG scsi_accel_rp2040_finishRead(0x08 0x00 0x00 0x01 0x01 0x00 
) detected parity error

...

[113786ms] DBG ---- COMMAND: Write6
[113787ms] DBG ------ OUT: 0x0A 0x00 0x7E 0x07 0x01 0x00
[113787ms] DBG ------ Write 1x512 starting at 32263
[113788ms] DBG ---- DATA_OUT
[113876ms] DBG scsi_accel_rp2040_finishRead(0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 ... (total 512)) detected parity error
[113879ms] DBG scsi_accel_rp2040_finishRead() detected parity error
[113879ms] DBG ---- Total IN: 0 OUT: 512 CHECKSUM: 0
[113880ms] DBG ---- STATUS: 2 CHECK_CONDITION, sense 0x00004700
[113888ms] DBG ---- MESSAGE_IN
[113888ms] DBG ------ IN: 0x00
[113890ms] DBG -- BUS_FREE
[113894ms] DBG ---- SELECTION: 1
[114045ms] DBG scsi_accel_rp2040_finishRead(0x03 0x00 0x00 0x00 0x00 
0x00 ) detected parity error

Peter

(Issue #27)

mynameistroy, 2023-04-30:
Alrighty, thanks for the clarification!  So definitely still a bug and I can test it on a MegaSTE which is very dependent on parity being disabled.

I'll look into it and hopefully have a fix in place soon.

mynameistroy, 2023-04-30:
I have an experimental build for you to try if you don't mind. The error message is still printed in the log but the error is ignored as expected. I've test it on my MSTE which requires disabled parity to function.

https://drive.google.com/file/d/1jUg6Oe0Ve-gWagrlIb5mWyBWLB1wHTXi/view?usp=share_link

mynameistroy, 2023-05-01:
I suspect the fix itself will be in shortly however I don't believe in removing the error from being displayed at this time. As it is a parity error, it's just being chosen to ignore it at the time.

mynameistroy, 2023-05-02:
PR #30 Is now ready 

Thanks again for finding and testing it out Peter.

z80micro-mc, 2023-05-02:
Troy,

I would prefer the error log to reflect the actual. However, as you 
prefer to report the parity error.  When the reporting is updated I 
sugest that the log entry should change to say 'Parity violation 
detected but ignored' when Parity=0

Peter

(Issue #27)

closed 2023-05-02 by mynameistroy

llemaire1, 2023-09-03:
Hello,

I'm trying to use a BlueSCSI card to emulate my NED Synclavier II hard-disk.

The only "special" need from the Synclavier, is to have SCSI drives with Parity Disable Checking.

I have upgraded my BlueSCSI module with the last firmware available (11th august 2023).

Here is my bluescsi.ini file content :

Debug=1

[SCSI5]
BlockSize=512
EnableParity=0
Parity=0
EnableSCSI2=0
Type=0

========================================================================================

Here is the log.txt output :

[10ms] Platform: BlueSCSI Pico
[10ms] FW Version: 2023.08.10-release Aug 11 2023 01:25:40
Flash chip size: 2048 kB
 
=== SD Card Info ===
SD card detected, exFAT volume size: 29503 MB
SD Name: SD32G, MID: 0x27, OID: 0x50 0x48
 
=== Global Config ===
Reading configuration from bluescsi.ini
Active configuration:
 
=== Finding images in / ===
== Opening /HD50_512 Ramon.hda for ID: 5 LUN: 0
---- WARNING: This image does not appear to be a valid Macintosh Device image. See: https://github.com/BlueSCSI/BlueSCSI-v2/wiki/Disk-Images
---- Image ready

=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- ROM drive image not detected
 
=== Configured SCSI Devices ===
* ID: 5, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 81920kB
 
Initialization complete!
INFO: Pico Voltage: 3.185V.

========================================================================================

I have two questions...
- Why don't I see the debug informations in the log.txt file ? do I need a special firmware version ?
- The Synclavier make an error when booting on the BlueSCSI drive, which refers to a problem with the parity check ...

I'm not sure my bluescsi.ini file is taken into consideration.... how can I be sure it is read and all the options taken into account ?

By the way, the BlueSCSI card is recognized and works fine when connected to a Mac using a RATOC FR1SX adapter and using Synclavier InterchangeX software to read the image disk.

Thanks in advance for your help !

Best regards.

     Laurent.