BlueSCSI/BlueSCSI-v2 #28: How about adding drive emulations rather than sort of systems

z80micro-mc, 2023-04-28:


z80micro-mc, 2023-04-28:
How about adding drive emulations rather than sort of emulations?

ie Quantum Pro40S
    SEAGATE ST 225N

Plus options for each as per the config jumpers on the drives?

Peter

erichelgeson, 2023-04-29:
Ya, that's a good idea - if you could provide a few pre-sets I'd be happy to add them.

erichelgeson, 2023-04-30:
https://github.com/PiSCSI/piscsi/blob/ef49663202f935b6a797a576a3413b7612075860/python/web/src/drive_properties.json

Some examples in the PiSCSI repo - not sure what from this list would be useful to implement (or not)

z80micro-mc, 2023-04-30:
Eric,

The link looks interesting as a starting point. Which then goes back to 
one of my original questions.

How do I configure / specify BlueSCSI device capacity?   - The only 
mechanism I can see at present is to configure CHS (cylinders, heads, 
sectors) Which is an alien concept to SCSI being a block based device, 
though I do accept that ultimately drives have numbers of cylinders, 
heads and sectors per track. I would like to define 'size=xxxxxx' where 
xxxxxx is the number of blocks.

Thanks

Peter

rather than sort of systems (Issue #28)

https://github.com/PiSCSI/piscsi/blob/ef49663202f935b6a797a576a3413b7612075860/python/web/src/drive_properties.json

erichelgeson, 2023-04-30:
The size is determined by how large your drive image is, eg: a 1GB drive image is exposed as a 1GB SCSI drive. Changing the defaults for CHS is normally not required unless your system cares about it. Most don't, but some do.

z80micro-mc, 2023-04-30:
Hi Eric,

So just to confirm if I present BlueSCSI with a 20 mbyte file it will 
report the capacity as 40960 blocks, and a 40 mbyte file as 81920 
blocks?

So what do CHS settings affect?

Thanks

Peter

erichelgeson, 2023-04-30:
> So what do CHS settings affect?

If, for whatever reason, your system needs some other values. Just because the options exist in the ini files does not mean you should set them, in fact normally you should not need to modify anything. They are just there for times you need to - those cases are usually documented on the Compatibility wiki page.

erichelgeson, 2024-05-22:
If you have any specific devices to implement please open an issue for those and the use case. We've recently done the iomega zip100 for example due to some samplers requiring them. I want to avoid just making one without a specific use case though. I'll close this for now as it seems to have addressed your question on CHS, please let us know if you have any other questions.

closed 2024-05-22 by erichelgeson