BlueSCSI/BlueSCSI-v2 #29: List of CCS commands implemented in BlueSCSI plus any limitations

z80micro-mc, 2023-04-28:
Is there a published list of SCSI Common Command Set (CCS) comands as implemented by BlueSCSI? 

Or in particular those which are not implemented and those with only a partial implementation.

This would be far more use to those of us trying to get BlueSCSI running on non-MAC hardware.

Peter

erichelgeson, 2023-04-29:
I wrote this up quite a while ago, so may be outdated (I'll add a dev notes wiki page with this later). Clause sections are in https://www.staff.uni-mainz.de/tacke/scsi/SCSI2.html

I did not capture if some of them are no-ops, (as many are as we don't have the physical characteristics or ability to respond with anything meaningful.) 

If there's a particular command you're interested in it's best to just jump into the command handler code, they are all pretty straight forward.

Command | OpCode | Type | Clause | BlueSCSI v1 | BlueSCSI v2 | PiSCSI
-- | -- | -- | -- | -- | -- | --
CHANGE DEFINITION | 40h | Optional | 8.2.1 | N | N | N
COMPARE | 39h | Optional | 8.2.2 | N | N | N
COPY | 18h | Optional | 8.2.3 | N | N | N
COPY AND VERIFY | 3Ah | Optional | 8.2.4 | N | N | N
LOG SELECT | 4Ch | Optional | 8.2.6 | N | N | N
LOG SENSE | 4Dh | Optional | 8.2.7 | N | N | N
SEARCH DATA EQUAL | 31h | Optional | 9.2.14.1 | N | N | N
SEARCH DATA HIGH | 30h | Optional | 9.2.14.2 | N | N | N
SEARCH DATA LOW | 32h | Optional | 9.2.14.3 | N | N | N
SET LIMITS | 33h | Optional | 9.2.16 | N | N | N
WRITE SAME | 41h | Optional | 9.2.24 | N | N | N
REASSIGN BLOCKS | 07h | Optional | 9.2.10 | N | N | Y
WRITE LONG | 3Fh | Optional | 9.2.23 | N | N | Y
READ LONG | 3Eh | Optional | 9.2.9 | N | N | Y
READ TOC | 43h | ? |   | N | N | Y
GET EVENT STATUS NOTIFICATION | 4ah | ? |   | N | N | Y
READ 16 | 88h | ? |   | N | N | Y
WRITE 16 | 8ah | ? |   | N | N | Y
VERIFY 16 | 8fh | ? |   | N | N | Y
SYNC CACHE 16 | 91h | ? |   | N | N | Y
READ CAPACITY 16 | 9eh | ? |   | N | N | Y
REPORT LUNS | A0h | ? |   | N | N | Y
RECEIVE DIAGNOSTICS RESULTS | 1Ch | Optional  | 8.2.13 | N | Y | N
SEEK6 | 0Bh | Optional  | 9.2.15 | N | Y | N
SEEK10 | 2Bh | Optional  | 9.2.15 | N | Y | N
WRITE AND VERIFY | 2Eh | Optional  | 9.2.22 | N | Y | N
SCSI Write sector buffer? | 0Fh | ? |   | N | Y | N
FORMAT UNIT | 04h | Mandatory | 9.2.1 | N | Y | Y
RELEASE | 17h | Mandatory | 9.2.12 | N | Y | Y
RESERVE | 16h | Mandatory | 9.2.12 | N | Y | Y
READ DEFECT DATA | 37h | Optional | 9.2.8 | N | Y | Y
START STOP UNIT | 1Bh | Optional | 9.2.17 | N | Y | Y
LOCK-UNLOCK CACHE | 36h | Optional | 9.2.2 | Y | Y | N
PRE_FETCH | 34h | Optional | 9.2.3 | Y | Y | N
READ BUFFER | 3Ch | Optional | 9.2.12 | Y | Y | N
WRITE BUFFER | 3Bh | Optional | 8.2.17 | Y | Y | N
SYNCHRONIZE CACHE | 35h | Optional | 9.2.18 | Y | Y | Y
PREVENT-ALLOW MEDIUM REMOVAL | 1Eh | Optional | 9.2.4 | Y | Y | Y
SEND DIAGNOSTIC | 1Dh | Mandatory | 8.2.15 | Y | Y | Y
INQUIRY | 12h | Mandatory | 8.2.5 | Y | Y | Y
READ6 | 08h | Mandatory | 9.2.5 | Y | Y | Y
READ10 | 28h | Mandatory | 9.2.6 | Y | Y | Y
READ CAPACITY | 28h | Mandatory | 9.2.7 | Y | Y | Y
REQUEST SENSE | 03h | Mandatory | 8.2.14 | Y | Y | Y
TEST UNIT READY | 00h | Mandatory | 8.2.16 | Y | Y | Y
MODE SELECT6 | 15h | Optional | 8.2.9 | Y | Y | Y
MODE SELECT10 | 55h | Optional | 8.2.8 | Y | Y | Y
MODE SENSE6 | 1Ah | Optional | 8.2.10 | Y | Y | Y
MODE SENSE10 | 5Ah | Optional | 8.2.11 | Y | Y | Y
REZERO UNIT | 01h | Optional | 9.2.13 | Y | Y | Y
VERIFY | 2Fh | Optional | 9.2.19 | Y | Y | Y
WRITE6 | 0Ah | Optional | 9.2.20 | Y | Y | Y
WRITE10 | 2Ah | Optional | 9.2.21 | Y | Y | Y

erichelgeson, 2023-06-02:
If you have any more questions please feel free to ask, seems like this is resolved for now.

closed 2023-06-02 by erichelgeson