BlueSCSI/BlueSCSI-v2 #31: Low capacity SD cards not detected

z80micro-mc, 2023-05-02:
Unlike v1, a 2Gb micro SD card is nolonger detected as valid - 5 flashes.
I have treied half a dozen different low capacity devices including: SanDisk 2 Gb, unmarked 2Gb, unmarked 1 Gb. It also fails to recognise a FujiFilm 128 Mb SD card.

The micro SD cards are detected as valid in V1 board.

I have tried 2 x 16 Gbmicro SD,: Kingston 16 Gb, and Sandisk 16 Gb both seem to work.

Peter

z80micro-mc, 2023-05-02:
Kingston 4Gb class(4) micro SD works out of the packaging. (It comes as FAT32 formatted)

z80micro-mc, 2023-05-02:
> (Issue #31)
>
> Strange! I do happen to have a 2gb sandisk and can confirm the same
> behavior, even after a full overwrite and CHS adjustment in SD Card
> formatter - logs [10ms] Platform: BlueSCSI Pico[10ms] FW Version:
> 2023.04.10-dev Apr 22 2023 15:56:08Flash chip size: 2048
> kBSdioCard::readSector(0x00000001) failed:
> 5SdioCard::readSector(0x00000089) failed:
> 5SdioCard::readSector(0x00000001) failed:
> 5SdioCard::readSector(0x00000089) failed: 5SD card init failed,
> sdErrorCode: 5 sdErrorData: 0SdioCard::readSector(0x00000001) failed:
> 5SdioCard::readSector(0x00000089) failed: 5SDIO SD card error on line
> 413, error code 2SDIO SD card error on line 413, error code
> 2SdioCard::readSector(0x00000001) failed:
> 5SdioCard::readSector(0x00000089) failed:
> 5SdioCard::readSector(0x00000001) failed:
> 5SdioCard::readSector(0x00000089) failed:
> 5SdioCard::readSector(0x00000001) failed:
> 5SdioCard::readSector(0x00000089) failed:
> 5SdioCard::readSector(0x00000001) failed: 5```

Eric,

How did you enable the sdio debug o/p when V2 won't recognise the sd 
card so cannot read the Bluescsi.ini file?

Thanks

Peter

erichelgeson, 2023-05-02:
https://github.com/BlueSCSI/BlueSCSI-v2/wiki/Developers-(Advanced)#usb-serial-debug

erichelgeson, 2023-06-02:
This is resolved in 66c1e47d298c273faec8458317ec60f7f0bb889c and will be in the next release (or you can grab the [nightly](https://github.com/BlueSCSI/BlueSCSI-v2/releases/tag/latest) build of main to test

closed 2023-06-02 by erichelgeson