BlueSCSI/BlueSCSI-v2 #33: ROM disk - seems to be fixed at floppy1.4

z80micro-mc, 2023-05-03:
I am attempting to create an MSDOS ROM disk.

I have an image of a 720k  floppy which I have loaded onto the Pico.  However, when I set 'Type = 3', it is reported as Floppy1.4 in the log file.

How do I generate / support a 720k disk?

Thanks

Peter

erichelgeson, 2023-05-03:
Yes, looks like it's hard coded to a 1.4mb floppy - it should be possible to make a profile for a 720k floppy. Could you try to set it as a "removable" instead of a floppy? eg: start the file name with `RE` instead of `FD`.

I dont have any DOS computers with a SCSI card so I wont be able to test this (actually I do have a PC PCI card... have to see if i can put together a DOS machine)

erichelgeson, 2024-05-22:
Just following up if you had a chance to try `RE` of 720k vs a `FD` - there really isn't much difference on the implemenation side besides product/vendor/id - and the `FlexibleDiskDriveGeometry` (which could need adjusting, I'd have to dig deeper into the spec).

closed 2026-09-26 by erichelgeson