BlueSCSI/BlueSCSI-v2 #34: Random write errors -SDIO SD card error on line 413, error code 2

z80micro-mc, 2023-05-04:
I am getting occasional write errors when I copy a number of files.

log.txt
======
[10ms] Platform: BlueSCSI Pico
[10ms] FW Version: 2023.04.10-dev Apr 30 2023 15:09:12
Flash chip size: 2048 kB
 
=== SD Card Info ===
SD card detected, FAT32 volume size: 7615 MB
SD Name: ASTC, MID: 0xFE, OID: 0x34 0x56
 
=== Global Config ===
Reading configuration from bluescsi.ini
Active configuration (using system preset "Generic"):
-- SelectionDelay: 150
-- EnableSCSI2 is off
-- Parity is disabled
 
=== Finding images in / ===
== Opening /HD10_512.hda for ID: 1 LUN: 0
---- Image ready
== Opening /HD20_512.hda for ID: 2 LUN: 0
---- Image ready

=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- ROM drive SCSI id 1 is already in use, not enabling
 
=== Configured SCSI Devices ===
* ID: 1, BlockSize: 512, Type: Fixed, Quirks: None, Size: 31980kB
* ID: 2, BlockSize: 512, Type: Fixed, Quirks: None, Size: 63960kB
 
Initialization complete!
SDIO SD card error on line 413, error code 2
SDIO SD card error on line 413, error code 2
SDIO SD card error on line 413, error code 2
SDIO SD card error on line 413, error code 2
SDIO SD card error on line 413, error code 2
SdioCard::stopTransmission() timeout
SD card write failed: 0x00
SDIO SD card error on line 413, error code 2
SDIO SD card error on line 413, error code 2
SdioCard::stopTransmission() timeout
SD card write failed: 0x00
SDIO SD card error on line 413, error code 2

Thoughts / ideas please

Peter

Konf, 2023-05-04:
Hi! I had such errors with old and slow SD card. Replaced it with new a one and no such errors appear anymore.

z80micro-mc, 2023-05-04:
Konf,

Thanks for your input.

I am currently using an 8 GB, C10, U1 Kingston micro SD rated at 80 MB/s. I obtain ~ 180 KB/s read or write via DOS. The SD is nearly 200 times faster than the transfer speed. On a large number of writes (copy all the files on a disk to another disk) I intermittently get write errors ~ every once every 5-10 minutes, at the current transfer rates I estimate ~ 1 failure in ~ 50,000 write sector operations.

To my reasoning is that is rather indicative of a timing race condition. A faster SD may reduce the likelyhood of errors but I think that there is a fundamental timing race condition. I would be tempted to slow the SD right down to make the condition worse to investigate further.

To date I have only ever seen, errors on write and repeating the same write subsequently succeeds. I have never, seen a failure on read.  The write failure gets mapped to a general failure reported by DOS, if one selects the 'r' repeat operation it proceeds without error until the next random event.

Peter

z80micro-mc, 2023-05-04:
Eric,

If it would be helpful, I could put a logic analyser on the SCSI bus.  However, as the event is so infrequent, it would be helpful if there is a spare gpio bit which you could toggle on error. I could then trigger the analyser on the bit change.

The analyser is capturing all the time and can be set to stop on trigger event, well 25% buffer before or after the event selectable.

Peter

erichelgeson, 2023-05-06:
No free pins but you can re-purpose `SDBG`, `S1`, or `S2` since they are normally unused. Interested in any results you get.

z80micro-mc, 2023-05-06:
Eric,

OK dev version with debug error pulse on S1 / S2 pin please. It needs to 
have Troys mods in for parity disable.

Peter

error on line 413, error code 2 (Issue #34)

erichelgeson, 2023-05-06:
I had to use the `SDBG` (pin 16) - it will go high on an SD error. If you'd like a pulse (how long) or delay or something let me know. 
https://github.com/BlueSCSI/BlueSCSI-v2/actions/runs/4902832099 <-- binaries at the bottom (includes troy's patch)
https://github.com/BlueSCSI/BlueSCSI-v2/commit/227baab4b654d54f0d333320cdba79beb3886591 <-- change

z80micro-mc, 2023-05-06:
Eric,

That should be fine.

When is the bit cleared? Might I presume on the next valid operation

Peter

  https://github.com/BlueSCSI/BlueSCSI-v2/actions/runs/4902832099 
<https://github.com/BlueSCSI/BlueSCSI-v2/actions/runs/4902832099>  <-- 
  227baab

erichelgeson, 2023-05-06:
This build will clear the assertion on next success read or write - https://github.com/BlueSCSI/BlueSCSI-v2/actions/runs/4902950485

z80micro-mc, 2023-05-06:
Eric,

Many thanks for the debug version, I'll dig out the analyser tomorrow 
and see what I can capture.

Are there any particular signals that you would like to see in the 
capture?

I have 8, 16, 32, or 48 channels depending upon which analyser I use. 
The 8, 16 & 32 chanels I can get into Sigrok Pulseview so you can browse 
them your self. The 48 channel analyser is a Tek 1225, which is perhaps 
easier to use but has limited export capabilities.

Peter

https://github.com/BlueSCSI/BlueSCSI-v2/actions/runs/4902950485

erichelgeson, 2023-07-03:
Please test with the latest release and report back, there were various fixes related to SD cards.

erichelgeson, 2024-05-22:
Hi, any update or anything for us to follow up on? We've seen this a few times and usually a full overwrite format with SD formatter as mentioned takes care of it, or cleaning the contacts on the SD card (can get dirty!), or trying another SD card.

erichelgeson, 2024-09-15:
The latest release will likely will resole you're issue, please let us know! https://github.com/BlueSCSI/BlueSCSI-v2/releases/tag/v2024.09.15

closed 2024-09-15 by erichelgeson