BlueSCSI/BlueSCSI-v2 #37: DB25 External 3D Printed Case a little too flimsy

armadillu, 2023-05-10:
I just got an external DB25 BlueScsi V2, and I found the 3D-printed case available in this repo a little too thin for my liking. I made a sturdier version based on the original stl files which works for me. This version keeps the idea of clickable buttons (which I found very neat), and also added little holes so you can see LED status. It is sturdier than the original, and is generally a little more polished. Happy to make a PR if you like it.

https://www.youtube.com/watch?v=jpjAizllZkU

erichelgeson, 2023-05-11:
Thanks! I've said it before but we're not 3D designers so appreciate any community cases/modifications. For now how about you publish the updated STL to a repo under you're user and link it in the Accessories Wiki page? We do actually have an openscad file for this and would like to publish  it so people can do some customization in the future.

armadillu, 2023-05-11:
Added to the wiki, and fixed the existing link for the Stock DB25 case as it was pointing to the wrong place

closed 2023-05-11 by armadillu