BlueSCSI/BlueSCSI-v2 #38: A better (or less crappy) intruction set?

thetylander, 2023-05-18:
I have been fighting (literally) for days to get the POS Blue Scummy to work on a Macintosh Plus. Compared to just about anything else, this is the ultimate in half assed garbage. 
Funny how you can plug in a Floppy Emu and it just works... 
This isa high profit low quality joke, 
Just to answer the "I'm stupid and we who made it are smart" questions.

Yes, I purchased from an authorized scammer. 
     eBay ID: jwg1 - It is a very crappy POS and has no return policy. A waste of $65.00

I downloaded ever single piece of shovelware listed on every page and updated the pico. This isn't my first rodeo.

I attempted to use the Blue Scummy v2 with 1MB, 2 MB, and two different sets of 4MB of ram just to see if it made a difference.
I also used two motherboards, an early 1986 board and a board from 1989 which was a factory 4MB version.

yes, I created the scummy ini file.

Anyone remember the old Mac commercial where they show an IBM and say "In order to use this, you'll need to read this" and a stack of three ring binders hit the table. Then they show the Macintosh and drop a little pamphlet,
Sad how the Blue Scummy  idiots complicated the most simple machine. 

I know I have no right to be angry, and should join the discord for tech support and a plethora of dick and fart jokes.
You shouldn't have to join a social media cult site to get answers. They all should be available and accessible to everyone who buys this POS. 
ALONG WITH A TROUBLESHOOTING SECTION THAT DOESN'T TELL YOU TO FORMAT YOUR MEMORY CARD USING A SHOVELWARE PROGRAM OR BUY A FASTER MEMORY CARD!
 I currently have seven NEW SD cards PNY, Sandisk, etc. with varying sizes and speeds, All formatted using the slow shovelware program several times after each failure.

Even the hda files I have used boot nicely on emulators. 

what do we get? Sometimes its blinking oncesometimes twice, sometimes three times and so on.
Would it kill anyone to tell us what each of these blinking LED codes mean?
I guess it would. 
In the you can add a rom file to your Pico section., it tells you where blanks are, and tells you you can include start up files in the rom, but it is locked once it is copied to the pico...
soooooooo...
how the hell do you put accessible files on there and what files will actually  work or even boot the system?

ANYONE WANT TO ANSWER THAT?

No?

I opened an empty one without proper instruction from this github page using ciderpress and copied an System disk in. I can see it once I boot up using a floppy disk but it says it can't boot from the rom!!!!!
HAHAHAHAHAHAHAHAHA!
Did I mention that the Macintosh was the easiest computer? So why have people like you ruined it with shovelware, files that don't work  and vague instructions?
I just want a SCSI hard drive to boot from and store files on. Not to be laughed at while I swear and struggle with this POS. 

Oh well,.
Delete this and banish me from github for hurting your feels if you like. The truth is that Shit creating shit is still shit. Enjoy my $60 and the good laugh at my utter hopeless frustration, After all, it's funny isn't it! 
"SOMEONE ACTUALLY BOUGHT THIS CRAP AND THOUGHT IT WOULD WORK! HA HA HA HA HA!"

I look at my Macintosh Plus and seriously consider smashing it with a hammer and chucking it in the dumpster, I no longer find retro computing fun thanks to Blue Scummy and this whole ordeal.

RonsCompVids, 2023-05-18:
@thetylander 

I'm a long time user of the BlueSCSI - and have even used it a time or two with my Mac Plus.  A little while back I produced a video to try and assist folks who were having troubles with the official setup guide (especially folks who might not have a second Mac to use for some of the heavy lifting of getting an image created).

https://www.youtube.com/watch?v=szXAcwyee9g

I've gotten good feedback from other viewers who've found it useful.  I plan to make an updated one in the next week or two that incorporates several new features of the BlueSCSI2.

Give it a view, and if there are any questions I can answer I'll be happy to help if I can.

I hope you have a great day,

Ron

erichelgeson, 2023-05-19:
First - be respectful in your communications here. It's ok to be frustrated when you're faced with issues, but we need to be respectful when communicating.

Second thanks @RonsCompVids for the helpful video on making disk images. This is the kind of help and friendliness you can expect from the community.

I'll sift through and answer your questions:

> and should join the discord for tech support 

You do not need to join Discord for tech support. You can always open an issue on github (like this one) or contact the seller (Jay is a great guy, he always goes above and beyond helping. He has no record of you contacting him)

> to work on a Macintosh Plus

As you saw in our docs the Mac Plus requires a special configuration (it and the Portable being the only Macintosh ones that do) as they were very early and have quirks in the way they implemented SCSI. Do I wish there was a way to make it "just work" with a Plus at the same time as every other machine? Sure, but currently there is not.

> Even the hda files I have used boot nicely on emulators.

This is not an indication that the image you created will work in a BlueSCSI. Most emulators default to creating a HFS volume image, which is not what is on a real SCSI disk. We provide many pre-made images, pre-made blanks, and Disk Jockey to get you going. See the Usage page.

>  ALONG WITH A TROUBLESHOOTING SECTION THAT DOESN'T TELL YOU TO FORMAT YOUR MEMORY CARD USING A SHOVELWARE PROGRAM OR BUY A FASTER MEMORY CARD!

The SD Card Associations SD Card Formatter resolves so many issues with odd issues with formatting. The creator of the opensource SDFat library we use to access SD Cards recommends it too. It's a great tool and fixes many issues. Some old memory cards are very slow and will cause issues if you use them on a Fast SCSI bus, hence the recommendation.

> Would it kill anyone to tell us what each of these blinking LED codes mean?

They are listed on the troubleshooting page. I've also added some details to make it more clear. It sounds like yours was flashing due to SCSI bus activity (looking for a boot disk, not finding one due to putting only an HFS volume image on) not any error code. I added a Mac specific section to the page to address this.

> I opened an empty one without proper instruction from this github page using ciderpress and copied an System disk in. I can see it once I boot up using a floppy disk but it says it can't boot from the rom!!!!!

You again likely created a HFS volume instead of using the provided disk images.

> Not to be laughed at while I swear and struggle with this POS.

We're all pretty friendly on here. It's OK to ask for help, respectfully.

If you would like to continue to troubleshoot you can continue here with respect for others. If not I'll close the issue and you can move on.

thetylander, 2023-05-19:
I was an angry frustrated idiot. 
Again, it really shouldn’t be this difficult, even setting up a retropie is easier. 

At the peak I had twelve hard disk images formatted, from blank to basic start up. Yes, all of them labeled and formatted correctly. HD20_512.hda for example. I downloaded premade ones from Macintosh Garden (rascsi and piscsi) The Internet Archive, and so on. 

I had the right files, I set it up according to instructions, updated the firmware, formatted several new SD cards, installed the .ini file, used three and a half different Mac Plus configurations and two motherboards. 
And multiple SD card configurations. Too many to list here. 
The log never showed an error, except when I did it on purpose to see if it was lying! It wasn’t. According to the log everything was super. 
Even LIDO wouldn’t show a scsi chain. 

The clue was when I gave up after typing my angry message last night and played several vigorous rounds of MacMan off of a genuine floppy disk. 
I was trying desperately to love my Mac again. 
I carelessly left the Blue SCSI attached, and when I quit the game, HD40 6.0.8 was on the desktop. 
It was corrupt though and needed reformatting. 

I did it right on the system, reinstalled 6.0.8 off of floppy disks and restarted. 
The Hard drive image was gone again.

I never thought about questioning the Blue SCSI hardware build quality. I assumed it was good. After all, the seller says he tests them before sending them out, thus his reason for his zero return policy. 
My teenage son actually pointed it out. 
I work on a lot of pinball machines and he said “You never question the proms, only the hardware on the motherboards.” 

I slapped my forehead and began inspecting for missing components or bad solder joints. 
It was the latter. Right on the Blue SCSI pin header. Just like an Williams system 11 pinball MPU with cold solder joints, sometimes it boots up, but mostly it doesn’t.

The most frustrating thing was that this kind of discovery only happens after you make an ass of yourself. 
Which is typical, and I do apologize. 

Not that there couldn’t be much better documentation. I’d like to see it where even stupid jerks like me who start swearing after multiple days of frustration can figure it out.
Everyone says it’s hard to get it working on a Mac Plus. I got lucky and didn’t know it. 

BTW: YouTube Blue SCSI people: that’s a step by step video idea! Do a fresh install and activation on a Mac Plus. There are zero videos on the subject. 

The flashing light indicators should all be in a section. I’m only guessing but one blink seems to indicate there’s no Sd card. Two indicates a HD file is present and three seemed to only happen when I had the blank rom file on the Pico and a HD hda on a memory card. 

Again, I work on pinball machines, so if you get to three blinks and then nothing on an old Bally, for example, it means your ram is bad or not detected. Later pinball machines yell at you what’s wrong on the DMD. I blindly assumed  the flashing lights on the Blue SCSI would drop error codes, not just indicate that it’s working. It can’t detect that the data is not going to the computer because of a bad solder joint, it just shows that the thing is on and the software is on it. 

How to install software on a bootable blank rom file for the Pico is another thing I never found out how to do. 
This should be explained better in the setup guide. All I do know is once the rom is on there, it can’t be written to and so you have to blank it off and try again after getting a sad Mac. 

Anyway, bad Blue SCSI solder job repaired and it finally freakin worked. 
And I’m still shaking my head every time I think about the last four and a half days. 

I think the only thing I’d like to say is that I got a nice lecture about treating people with respect, regardless of if they deserve it or not. 
I come from an era where respect is earned, not freely given. I know that I did not earn respect for losing my temper. 
Nor did I expect it to be freely given to me. 
I was a very dissatisfied customer who placed trust and my hard earned dollar bills in someone’s hand. The trust and respect tend to go out the window when you pay for a product that doesn’t live up to expectations. 
And the whole time I really thought it had to be my fault for doing it wrong. 
I was a jerk, but it took me four days to become one. 

On the incredibly rare occasion I mess up a customer’s pinball, I expect them to swear and yell, and they do. Then I fix it, give them a partial refund, and apologize. I don’t expect or demand they respect me for doing a half assed job. 

On that topic, even though it turned out to be a hardware fault, I’ll be leaving the fleabay seller positive feedback because bad solder sometimes happens.  

The software and setup side is a mess though. Perhaps premade zip files with a basic working system setup that can be dragged and dropped on to an SD card for specific machines and operating system versions would be an idea. 
It would be kind of like those start up disks you got with a new Mac back in the day. After all, the Macintosh was the computer for the rest of us. 
It would then be up to the user to populate the hard drive image with utilities, apps and game software as they see fit. 
it wouldn’t be hard to do, and would save a lot of trips from one download site to the next to the next to the next. 
It’s just an idea.

erichelgeson, 2023-05-23:
Very glad you got it resolved. I've done some updates to the documentation based on your feedback. Thanks.