BlueSCSI/BlueSCSI-v2 #40: Some Toast CD images dont mount correctly?

armadillu, 2023-05-20:
I have a couple particular Toast images that mount in a strange way when used through BlueScsi V2 (v2023.04.10). The same exact file mounts ok when copied into the hard drive and mounted through the OS (either with toast, Virtual CD/DVD, etc). When mounted this way, the os gives a read error when trying to access the files. It seems to be an issue with the particular file; it always happens with the same files.

This is an example of one such files: https://macintoshgarden.org/sites/macintoshgarden.org/files/games/TheDig_1_0.sit

You can tell right away, when mounted through blueSCSI the mounted CD shows no icons, but when mounted directly through the OS, app icons show correctly. The issue happens on macos 7.5.3, 7.6, 8.1, 8.6 and 9.1 on my machine (a color classic with a takky 5300/100 applied).

There's nothing relevant I can see on the log.txt file, and no error file is created.

[image]

[image]

erichelgeson, 2023-05-20:
Not all toast images are CD images - some are just a HFS volume. It does not have the actual bytes that are on a CD, which BlueSCSI would require (as we just read bytes off the sd card), but tools like such as image mounters/etc can mount. I'll take a deeper look later this weekend but that would be my guess.

armadillu, 2023-05-20:
Yep that makes sense, and I can confirm. I compared some of the toast images that work with BlueSCSI with the ones that don't, and you can see that all the working ones start with magic bytes "ER"... While the other ones start with zeros.

[image]

`(On the left, a working toast image. On the right, a non-working one)`