BlueSCSI/BlueSCSI-v2 #42: forgot to flash the pico before soldered it to the bluescsi v2 board

maded2, 2023-05-31:
hi,
  forgot to flash the pico before soldered it to the bluescsi v2 board.  But now I can't get it to go into flashing mode by holding down the boot_sel.  Is there any work around? or do I have to desolder the pico?

thanks in advance
eddie

erichelgeson, 2023-05-31:
You do not need to remove the pico to flash. Make sure you're using a USB cable that is power only. 

Hold down boot sel, connect to a computer via USB, copy the .uf2 file to the drive that shows up.

https://github.com/BlueSCSI/BlueSCSI-v2/wiki/Updating-Firmware#usb-method

maded2, 2023-05-31:
If I hold boot_sel and plug the usb into the PC, nothing happens, it doesn't go into the usb drive mode.

androda, 2023-05-31:
Does this microUSB cable work for data transfer on other devices?  It sounds like you have a power-only cable.  I regularly reprogram soldered-down pico modules via USB, it should work fine.

maded2, 2023-05-31:
I usef the same cable to flash another pico to make sure it worked.  So the cable is fine. I guessed i will desolder it tomorrow

erichelgeson, 2023-06-02:
Did you figure it out? Maybe a solder ball or something got where it shouldn't.

maded2, 2023-06-03:
Had to desolder it from the board.  Then I can get the pico into flashing mode.  I did try to power the board and flash, but still no good.