BlueSCSI/BlueSCSI-v2 #45: Setting block size on CDROM sets

henris42, 2023-06-21:
I'm trying to get a latest release v2 working with a SGI Indigo R4400.

For installing operating system 2 CDs are required so I need disk set. However the block size of SGI CDROM is 512, so it cannot boot from the CD.

Is there any undocumented way to set block size for a set?

henris42, 2023-06-22:
Thanks, I'll recompile for the time being. Hopefully there will be a new config param in the next release.

erichelgeson, 2023-06-22:
I don't think we need a new config param (there are WAY too many as it is :D ) but we should be honor the block size if specified in the file name or if there is a `BlockSize` key at the [SCSI#] config block level.

henris42, 2023-06-22:
OK, sounds good.

I tried to change the 2048 to 512 as suggested, but log still reports 2048:
--
```
[10ms] Platform: BlueSCSI Pico
[10ms] FW Version: 2023.04.10-hs Jun 22 2023 17:55:10
Flash chip size: 2048 kB
SDIO SD card error on line 422, error code 2
SDIO SD card error on line 422, error code 2
SDIO SD card error on line 422, error code 2
SDIO SD card error on line 422, error code 2
SD card init failed, sdErrorCode: 2 sdErrorData: 0
SD card init succeeded after retry
 
=== SD Card Info ===
SD card detected, exFAT volume size: 30551 MB
SD Name: JB1Q5, MID: 0x1B, OID: 0x53 0x4D
[14524ms]  
[14524ms] === Global Config ===
[14525ms] Reading configuration from bluescsi.ini
[14525ms] Active configuration (using system preset "Generic"):
[14526ms] DBG -- SelectionDelay: 255
[14526ms] DBG -- EnableUnitAttention is off
[14527ms] DBG -- EnableSCSI2 is on
[14527ms] DBG -- EnableSelLatch is off
[14528ms] DBG -- MapLunsToIDs is off
[14528ms] -- Debug is enabled
[14528ms] DBG -- Parity is enabled
[14529ms] DBG -- ReinsertCDOnInquiry is disabled
[14533ms] -- Opening '/CDs/IRIX 6.2 CD1.iso' for id:3, specified in bluescsi.ini
[14535ms] DBG ---- Read prefetch enabled: 8192 bytes
[14539ms]  
[14539ms] === Finding images in / ===
[14540ms] == Opening /HD10_512.hda for ID: 1 LUN: 0
[14541ms] DBG ---- Read prefetch enabled: 8192 bytes
[14541ms] ---- Image ready
[14542ms] 
[14542ms] === ROM Drive ===
[14543ms] Platform supports ROM drive up to 1692 kB
[14543ms] ---- ROM drive image not detected
[14543ms]  
[14544ms] === Configured SCSI Devices ===
[14544ms] * ID: 1, BlockSize: 512, Type: Fixed, Quirks: None, Size: 28672000kB
[14545ms] * ID: 3, BlockSize: 2048, Type: Optical, Quirks: None, Size: 557924kB
[14647ms]  
[14647ms] Initialization complete!
[14692ms] INFO: Pico Voltage: 3.326V.
[14694ms] DBG -- BUS_BUSY
```

erichelgeson, 2023-06-22:
Can you show your `bluescsi.ini`? Do you have block size set since you are not using the naming convention in the CD name? The hard coded value was only when cd's are switched, not when initially loaded.

henris42, 2023-06-22:
Its currently like this:

```
[SCSI]
System=Generic
Debug=1
Quirks=0

[SCSI3]
Type=2 # CD-ROM
BlockSize=512  
IMG0=/CDs/IRIX 6.2 CD1.iso
IMG1=/CDs/IRIX 6.2 CD2.iso
```

erichelgeson, 2023-06-22:
Found yet another hard coded value https://github.com/BlueSCSI/BlueSCSI-v2/blob/main/src/BlueSCSI_disk.cpp#L718

Artifacts from the upstream project - we'll get it fixed. Thanks for reporting.

henris42, 2023-06-23:
Yes, with that fixed the log looks now ok.

commit c8f223cd (2023-07-03): Fixes hard coded block sizes in multiple locations.
Adds BlockSize ini param and new default bytesPerSector preset.
Fixes #45
PR #47 (2023-07-03): Fixes hard coded block sizes in multiple locations.
Adds BlockSize ini param and new default bytesPerSector preset. Fixes #45
commit d23def1f (2023-08-09): Fixes hard coded block sizes in multiple locations.
Adds BlockSize ini param and new default bytesPerSector preset.
Fixes #45
closed 2023-07-03 by erichelgeson