BlueSCSI/BlueSCSI-v2 #62: Networking not working

patdie421, 2023-08-16:
Hi,

I am currently trying to get the Ethernet functionality of the latest firmware (v2023.08.10) to work with the following configuration file:

**BlueSCSI.ini**
```
[SCSI]
WiFiSSID=france2
WiFiPassword=********
# Optional: If you have more than one BlueSCSI WiFi connected to
# the same network you will have to change your MAC address:
 
# WiFiMACAddress=01:23:45:67:89 # Default is 00:80:19:C0:FF:EE
WiFiMACAddress=00:01:45:89:67:20
```

After starting, MacOS 7.6.1 french + Opentrasport 1.3 can see the network interface, but the Mac (IIsi) can't communicate on the network (DHCP or static IP).
The log file from BlueSCSI is:

**log.txt**
```
[12ms] Platform: BlueSCSI Pico
[12ms] FW Version: 2023.07.08-dev Aug 10 2023 17:49:40
Flash chip size: 2048 kB

=== SD Card Info ===
SD card detected, exFAT volume size: 15189 MB
SD Name: SC16G, MID: 0x03, OID: 0x53 0x44

=== Global Config ===
Reading configuration from bluescsi.ini
Active configuration:

=== Finding images in / ===
== Opening /HD60-1GB-exchange.dsk for ID: 6 LUN: 0
---- WARNING: This drive contains the LIDO driver and may cause issues.
---- Image ready
== Opening /HD10-2GB.hda for ID: 1 LUN: 0
---- WARNING: This drive contains the LIDO driver and may cause issues.
---- Image ready
== Opening /HD00 - System-7.1-mini.dsk for ID: 0 LUN: 0
---- WARNING: This drive contains the LIDO driver and may cause issues.
---- Image ready
== Opening /HD20 - Sytem-753FR.dsk for ID: 2 LUN: 0
---- WARNING: This drive contains the LIDO driver and may cause issues.
---- Image ready
== Opening /NE4.hda for ID: 4 LUN: 0
---- Configuring as network based on image name
---- Image ready

=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- ROM drive image not detected

=== Configured SCSI Devices ===
* ID: 0, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 102400kB
* ID: 1, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 2097216kB
* ID: 2, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 512000kB
* ID: 4, Type: Network, Quirks: Apple
* ID: 6, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 1048576kB

=== Network Initialization
Wi-Fi MAC: 00:00:00:00:01:45
WARNING: Wi-Fi MAC is not what was requested (%02x:%02x:%02x:%02x:%02x:%02x), is libpico not compiled with CYW43_USE_OTP_MAC=0?016913710332
Connecting to Wi-Fi SSID "france2" with WPA/WPA2 PSK

Initialization complete!
INFO: Pico Voltage: 3.265V.
Successfully connected to Wi-Fi SSID "france2"
```

On the computer, viewed TattleTech, the MAC address is good ($000145896720).

Same issue (same log.txt replace just "Wi-Fi MAC: 00:00:00:00:01:45" by "Wi-Fi MAC: 00:00:00:00::80:19") with the default MAC address.

Something to do on my side or is it a bug ?

Thank for help

Patrice

erichelgeson, 2023-08-16:
Could you try not setting the MAC address at all (leave it completely out of the ini)? It is optional (thought there may be a bug on setting it if the default works for you.

patdie421, 2023-08-16:
Hi Eric,

I just downloaded and reinstalledl the firmware but now I use version:

2023.07.08-dev Aug 10 2023 15:51:08

With this version (a little older) and the same bluesci.ini I haven't the warning:
```
[12ms] Platform: BlueSCSI Pico
[12ms] FW Version: 2023.07.08-dev Aug 10 2023 15:51:08
Flash chip size: 2048 kB
 
=== SD Card Info ===
SD card detected, exFAT volume size: 15189 MB
SD Name: SC16G, MID: 0x03, OID: 0x53 0x44
 
=== Global Config ===
Reading configuration from bluescsi.ini
Active configuration:
 
=== Finding images in / ===
== Opening /HD60-1GB-exchange.dsk for ID: 6 LUN: 0
---- WARNING: This drive contains the LIDO driver and may cause issues.
---- Image ready
== Opening /HD10-2GB.hda for ID: 1 LUN: 0
---- WARNING: This drive contains the LIDO driver and may cause issues.
---- Image ready
== Opening /HD00 - System-7.1-mini.dsk for ID: 0 LUN: 0
---- WARNING: This drive contains the LIDO driver and may cause issues.
---- Image ready
== Opening /HD20 - Sytem-753FR.dsk for ID: 2 LUN: 0
---- WARNING: This drive contains the LIDO driver and may cause issues.
---- Image ready
== Opening /NE4.hda for ID: 4 LUN: 0
---- Configuring as network based on image name
---- Image ready

=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- ROM drive image not detected
 
=== Configured SCSI Devices ===
* ID: 0, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 102400kB
* ID: 1, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 2097216kB
* ID: 2, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 512000kB
* ID: 4, Type: Network, Quirks: Apple
* ID: 6, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 1048576kB
 
=== Network Initialization ===
Wi-Fi MAC: 00:01:45:89:67:20
Connecting to Wi-Fi SSID "france2" with WPA/WPA2 PSK
 
Initialization complete!
INFO: Pico Voltage: 3.303V.
Successfully connected to Wi-Fi SSID "france2"
```

But for now the network doesn't work better :-(

Thanks for your help

Patrice

erichelgeson, 2023-08-16:
Could you try my first suggestion of removing the MAC address line from the ini file and trying again? In the TCP/IP control panel you have "Alternate Ethernet" selected?

patdie421, 2023-08-16:
I had already done the test (my first try before add the custom MAC address) and had the same warning:

That's the log with no "WiFiMACAddress" in the ini file.
```
[12ms] Platform: BlueSCSI Pico
[12ms] FW Version: 2023.07.08-dev Aug 10 2023 17:49:40
Flash chip size: 2048 kB

=== SD Card Info ===
SD card detected, exFAT volume size: 15189 MB
SD Name: SC16G, MID: 0x03, OID: 0x53 0x44

=== Global Config ===
Reading configuration from bluescsi.ini
Active configuration:

=== Finding images in / ===
== Opening /HD60-1GB-exchange.dsk for ID: 6 LUN: 0
---- WARNING: This drive contains the LIDO driver and may cause issues.
---- Image ready
== Opening /HD10-2GB.hda for ID: 1 LUN: 0
---- WARNING: This drive contains the LIDO driver and may cause issues.
---- Image ready
== Opening /HD00 - System-7.1-mini.dsk for ID: 0 LUN: 0
---- WARNING: This drive contains the LIDO driver and may cause issues.
---- Image ready
== Opening /HD20 - Sytem-753FR.dsk for ID: 2 LUN: 0
---- WARNING: This drive contains the LIDO driver and may cause issues.
---- Image ready
== Opening /NE4.hda for ID: 4 LUN: 0
---- Configuring as network based on image name
---- Image ready

=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- ROM drive image not detected

=== Configured SCSI Devices ===
* ID: 0, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 102400kB
* ID: 1, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 2097216kB
* ID: 2, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 512000kB
* ID: 4, Type: Network, Quirks: Apple
* ID: 6, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 1048576kB

=== Network Initialization
Wi-Fi MAC: 00:00:00:00::80:19
WARNING: Wi-Fi MAC is not what was requested (%02x:%02x:%02x:%02x:%02x:%02x), is libpico not compiled with CYW43_USE_OTP_MAC=0?016913710332
Connecting to Wi-Fi SSID "france2" with WPA/WPA2 PSK

Initialization complete!
INFO: Pico Voltage: 3.265V.
Successfully connected to Wi-Fi SSID "france2"
```

And yes, had "Alternate Ethernet" selected in TCP/IP.

I'll create a new fresh installation of System 7.1 to try with MacTCP. I'm not sure if my current OS (7.6.1 + OpenTransport 1.3) is very clean.

Patrice

patdie421, 2023-08-16:
After a fresh install of System 7.5.3 (haven't installation disks for 7.1) it's working with MacTCP and blueSCSI "2023.07.08-dev Aug 10 2023 15:51:08" firmware.
Tomorrow I'll retest with version "2023.07.08-dev Aug 10 2023 17:49:40" and Open Transport.
Thanks for your help.

Patrice

erichelgeson, 2023-08-17:
Glad it's working! I know people have got it working with OpenTransport. I would focus on only testing the latest version linked in the wifi github issue,

A side note, your log shows all your drives are using LIDO drivers, which is known to cause slow performance and other issues. You can use [Disk Jockey ](https://diskjockey.onegeekarmy.eu/) to swap out the drivers.

patdie421, 2023-08-22:
Hello,
Here is an update for this issue.
I managed to get the network working with MacIP and OpenTransport (1.3).
Problems came from my installations of MacOS. I don't know exactly why, but I guess it's because I created the images while installing with Basilist II. After reinstalling a "minimum" OS and following the Wiki installation procedure, I can have a functional IP connection, including to the Internet. I've done this tests on MacOS 7.5.3 and MacOS 7.6.1 with MacIP and Opentransport.
Just one more question, can Ethertalk be used (with a airport extreme Base Station 2003 who should be Ethertalk compible) with this network card emulation ?

PS: Thanks also for you suggestion for the scsi driver. Disk Jockey was used to remove LIDO driver with succes.

erichelgeson, 2023-10-07:
Do apologize for the delay and glad you got it working!

>  Just one more question, can Ethertalk be used (with a airport extreme Base Station 2003 who should be Ethertalk compatible) with this network card emulation ?

You can do IP Appletalk for sure, and I believe if your router is capable of passing along the packets. If you test it and it works (or doesn't) please let us know!

closed 2023-10-07 by erichelgeson