BlueSCSI/BlueSCSI-v2 #64: When using the CDX folder, disk initialization fails

aaronjmajors, 2023-08-29:
When using the CDX folder with a set of disk images, initialization fails with the following error (3x):

`WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)`

erichelgeson, 2023-08-30:
Can you provide more details?
Provide the full log.txt and err.txt if present
Provide a link to the disk images if possible
Does this happen on the latest release? (remove beta from the equation) 

Does this warning actually cause the CD to not work?

If you use all of these disk individually (eg the "normal" way by naming them CD3 .. etc) do you get the warning/do they work?

FWIW I've been using this CDX method exclusively since I wrote it and haven't had an issue. My best guess is one of the disks may not have proper padding (is common).

aaronjmajors, 2023-09-03:
Should be able to look at it again today or tomorrow, sorry for the delay!

aaronjmajors, 2023-09-04:
Ok here are a few more details... All of the disks eventually will mount after ejecting the first set. They're all showing up with an underscore ahead of their name, almost like hidden files (which aren't there) or an indexing issue. Full log below... in this instance I ejected all of the "failed to initialize" disks, then cycled through the ones that mounted, then it looped back to failing again...

`[12ms] Platform: BlueSCSI Pico
[12ms] FW Version: 2023.07.08-dev Aug 26 2023 21:35:31
Flash chip size: 2048 kB
 
=== SD Card Info ===
SD card detected, exFAT volume size: 30534 MB
SD Name: SD32G, MID: 0x28, OID: 0x42 0x45
 
=== Global Config ===
Reading configuration from bluescsi.ini
Active configuration:
-- SCSI ID: 2 using Optical image directory 'CD2'
-- Opening 'CD2/._Dig.toast' for ID:2
 
=== Finding images in / ===
== Opening /HD50_512 Apps 2000MB.dsk for ID: 5 LUN: 0
---- Image ready
== Opening /HD60_512 Macintosh LC III 80MB.dsk for ID: 6 LUN: 0
---- Image ready
== Opening /NE4.hda for ID: 4 LUN: 0
---- Configuring as network based on image name
---- Image ready

=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- ROM drive image not detected
 
=== Configured SCSI Devices ===
* ID: 2, BlockSize: 2048, Type: Optical, Quirks: Apple, Size: 4kB
* ID: 4, Type: Network, Quirks: Apple
* ID: 5, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 2048064kB
* ID: 6, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 81984kB
 
=== Network Initialization ===
Wi-Fi MAC: 00:80:19:08:43:45
Connecting to Wi-Fi SSID "XXXXXXXXX" with WPA/WPA2 PSK
 
Initialization complete!
INFO: Pico Voltage: 3.207V.
Successfully connected to Wi-Fi SSID "XXXXXXXXX"
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
Switching to next CD-ROM image for SCSI ID: 2: CD2/._Indy.toast
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
Switching to next CD-ROM image for SCSI ID: 2: CD2/._Marathon.iso
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
Switching to next CD-ROM image for SCSI ID: 2: CD2/._Myst.iso
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
Switching to next CD-ROM image for SCSI ID: 2: CD2/._Warcraft.toast
[... 41 lines cut ...]
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
Switching to next CD-ROM image for SCSI ID: 2: CD2/._Myst.iso
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
Switching to next CD-ROM image for SCSI ID: 2: CD2/._Warcraft.toast
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
[... cut ...]
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
WARNING: Host attempted read at sector 16+1, exceeding image size 2 sectors (2048B/sector)
Switching to next CD-ROM image for SCSI ID: 2: CD2/Dig.toast
`

erichelgeson, 2023-09-04:
It does appear there are hidden files there - on a Mac open the CD2 folder then press Command + Shift + . (the period key) to show the files. Remove the hidden ones. I will improve the cycling to skip hidden files.

erichelgeson, 2023-09-05:
Can you try this firmware and see if it resolves your issue? https://github.com/BlueSCSI/BlueSCSI-v2/suites/15855293814/artifacts/904712238

aaronjmajors, 2023-09-11:
Hey @erichelgeson - this version isn't initializing the files from my SD card at all it seems. Here's the log it's spitting out:

```
[10ms] Platform: BlueSCSI Pico
[10ms] FW Version: 2023.09.05-dev Sep  5 2023 21:28:02
Flash chip size: 2048 kB
 
=== SD Card Info ===
SD card detected, exFAT volume size: 30534 MB
SD Name: SD32G, MID: 0x28, OID: 0x42 0x45
 
=== Global Config ===
Reading configuration from bluescsi.ini
Active configuration:
-- 
```

erichelgeson, 2023-09-11:
Please turn on debug and show the exact contents of your SD card?

aaronjmajors, 2023-09-11:
Scratch that, it worked after I removed my `bluescsi.ini` file and also changed the name of my main boot drive from `.hda` to `.dsk`. Here's the full log:

```
[10ms] Platform: BlueSCSI Pico
[10ms] FW Version: 2023.09.05-dev Sep  5 2023 21:28:02
Flash chip size: 2048 kB
 
=== SD Card Info ===
SD card detected, exFAT volume size: 30534 MB
SD Name: SD32G, MID: 0x28, OID: 0x42 0x45
 
=== Global Config ===
Config file bluescsi.ini not found, using defaults
Active configuration:
-- SCSI ID: 2 using Optical image directory 'CD2'
-- Opening 'CD2/DarkForces.img' for ID:2
 
=== Finding images in / ===
== Opening /HD60_512 Power Macintosh 7200-90 500MB.dsk for ID: 6 LUN: 0
---- WARNING: file /HD60_512 Power Macintosh 7200-90 500MB.dsk is fragmented, see https://github.com/BlueSCSI/BlueSCSI-v2/wiki/Image-File-Fragmentation
---- Image ready
== Opening /HD50_512 Apps 2000MB.dsk for ID: 5 LUN: 0
---- Image ready

=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- ROM drive image not detected
 
=== Configured SCSI Devices ===
* ID: 2, BlockSize: 2048, Type: Optical, Quirks: Apple, Size: 76720kB
* ID: 5, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 2048064kB
* ID: 6, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 512064kB
 
Initialization complete!
INFO: Pico Voltage: 3.239V.
Switching to next CD-ROM image for SCSI ID: 2: CD2/Dig.toast
Switching to next CD-ROM image for SCSI ID: 2: CD2/Indy.toast
Switching to next CD-ROM image for SCSI ID: 2: CD2/Marathon.iso
Switching to next CD-ROM image for SCSI ID: 2: CD2/Myst.iso
```

erichelgeson, 2023-09-11:
File extensions are not considered when opening the drive (except for ignoring compressed files). Could you put it back to hda and retest. If it fails please include a debug log.

aaronjmajors, 2023-09-12:
Yup, looks like that worked as well for me... Thanks!

commit a758c656 (2023-09-12): Ignore hidden files. Fixes #64
commit d00ebe01 (2023-09-12): Merge pull request #68 from BlueSCSI/eric/64

Fix #64 - Issue with implict CD changing when hidden files are in dir
closed 2023-09-12 by erichelgeson