BlueSCSI/BlueSCSI-v2 #66: "Disable Parity Check" issue when using a BlueSCSI card on a NED Synclavier II.

llemaire1, 2023-09-03:
Hello,

I'm trying to use a BlueSCSI card to emulate my NED Synclavier II hard-disk.

The only "special" need from the Synclavier, is to have SCSI drives with Parity Disable Checking.

I have upgraded my BlueSCSI module with the last firmware available (11th august 2023).

Here is my bluescsi.ini file content :

Debug=1

[SCSI5]
BlockSize=512
EnableParity=0
Parity=0
EnableSCSI2=0
Type=0

========================================================================================

Here is the log.txt output :

[10ms] Platform: BlueSCSI Pico
[10ms] FW Version: 2023.08.10-release Aug 11 2023 01:25:40
Flash chip size: 2048 kB

=== SD Card Info ===
SD card detected, exFAT volume size: 29503 MB
SD Name: SD32G, MID: 0x27, OID: 0x50 0x48

=== Global Config ===
Reading configuration from bluescsi.ini
Active configuration:

=== Finding images in / ===
== Opening /HD50_512 Ramon.hda for ID: 5 LUN: 0
---- WARNING: This image does not appear to be a valid Macintosh Device image. See: https://github.com/BlueSCSI/BlueSCSI-v2/wiki/Disk-Images
---- Image ready

=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- ROM drive image not detected

=== Configured SCSI Devices ===

ID: 5, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 81920kB
Initialization complete!
INFO: Pico Voltage: 3.185V.

========================================================================================

I have two questions...

Why don't I see the debug informations in the log.txt file ? do I need a special firmware version ?
The Synclavier make an error when booting on the BlueSCSI drive, which refers to a problem with the parity check ...
I'm not sure my bluescsi.ini file is taken into consideration.... how can I be sure it is read and all the options taken into account ?

By the way, the BlueSCSI card is recognized and works fine when connected to a Mac using a RATOC FR1SX adapter and using Synclavier InterchangeX software to read the image disk.

Thanks in advance for your help !

Best regards.

 Laurent.

erichelgeson, 2023-09-03:
> Why don't I see the debug informations in the log.txt file ? do I need a special firmware version ?

Your `Debug` is not in any ini section - see ini.bluescsi.com or the Wiki for examples. eg:

EnableParity is only valid in the global `[SCSI]` section, you can not put it under a `[SCSIX]` section.

A cleaned up version of your config would be:

```ini
[SCSI]
Debug=1
EnableParity=0
EnableSCSI2=0
```

Note nothing under [SCSI5] is required as they are defaults or should be in the global section.

HTH, let me know if that works for you.

llemaire1, 2023-09-03:
Hello Eric,

Thanks so much for your help and the tips !!!   It works perfectly fine like that.

Maybe it could be interesting to write these infos in the bluescsi.ini wiki page ?

Again, thanks for your help !

     Laurent.

erichelgeson, 2023-09-05:
Glad it's working - I've updated the wiki page to show if it's global, device, or both for each ini item.

closed 2023-09-05 by erichelgeson