BlueSCSI/BlueSCSI-v2 #69: Attempt to use Spotlight index files

VitaliyYakob, 2023-09-22:
Hi,
Firmware v2023.08.10, created the CD2 directory when it placed the ISO images. macOS created Spotlight index files.
[image]

When change the image in a vintage Mac, an index file is inserted.

[image]

Of course, Mac OS can't read such a "image" of the disk

[image]

erichelgeson, 2023-09-22:
Hi! This is a dupe of #64 - There is a patch that will be in the next release. If you'd like to test it out click on "Releases" in github and download the Nightly one.

VitaliyYakob, 2023-09-22:
Oh, sorry. I didn't notice it.
Thanx.

closed 2023-09-22 by VitaliyYakob