BlueSCSI/BlueSCSI-v2 #73: Weird schematics

douardda, 2023-10-04:
Hi, 

thanks for this nice project (I have built myself and using a few BlueSCSI v1).

I was considering building a BlueSCSI v2, but unless I missed something, there are issues in the schematics. For example if I consider  the C50p version:

- R values are missing around CBTD3384 chips,
- Circuits around LVT245's are very weird. If I'm not mistaken, LVT245 is an octal bus transceiver, so A1 <-> B1 (etc) with DIR pin setting the direction of the buffer. In this schematics, we have:
  - DB0T <-> DB0 (ok could make sense)
  - GND-R11 <-> R45-GND : what is this supposed to do?
  - etc.
- The SCSI termination circuit does only link SCSI lines to the +2.8v via an unknown resistor, while the usual SCSI termination should be a pair of 220/330 Ω linked to vcc and GND. How does this schematics match the SCSI specs?
- The DB25 version seems not finished.

Are there improved/updated versions of these schematics that are waiting to be merged or something?

David

douardda, 2023-10-04:
thanks a lot for these explanations! These are worth comments in the schematics (I may send a PR; is it ok to propose revised versions of the kicad files made with kicad7 ? current files are kicad 6 I believe).

androda, 2023-10-05:
I'm currently preparing some hardware updates, and will update the schematics to be clearer.  I'm also planning to create a section on the wiki which calls out hardware features and design decisions (such as active termination).

douardda, 2023-10-05:
Ah nice. 

I also had in mind to split the schematics using more hierarchical sheets, and share some on them between the different versions of the device (C50, DB25, etc). For example all the core parts of the BlueSCSI are the same between all 4 of the versions (the rp2040, the lvt125's, some connectors and power rails & caps, etc.)

Kicad allows to use a sheet from another directory, so there is no unnecessary duplication.

androda, 2023-10-06:
That's an interesting feature.  Hadn't heard about it.  It's something to consider, although I don't personally see an issue with having fully individual schematics for each design.  They don't take up much space.

The designs are going to start diverging over time as they focus on different things.  Desktop 50 Pin will probably always have more features than the PowerBook version, as an example.  And the other external styles aren't exactly the same either.

I've opened #74 to clean up the DB25 schematic, and the other hardware versions will get cleanup as well as they are tested and released.

androda, 2023-10-13:
An updated Desktop schematic has a PR open now, and the C50 / PowerBook ones will follow next time they get a hardware update.

androda, 2024-05-22:
All released schematics have been cleaned up substantially, with comments and component values being more clear.

closed 2024-05-22 by androda