BlueSCSI/BlueSCSI-v2 #77: Updating Firmware with SD-Method doesn't work

explit7, 2023-10-07:
I have 2x BlueSCSIv2, One PowerBook Version, and one DB25 (Short).
Both devices just ignore the Firmware .bin File, when i put them to the root of the MicroSD-Card.

Can i somehow force the Firmware Update with a statement in the bluescsi.ini ?

While DB25-Version is quickly disassembled to update via the USB-Method, for PowerBook Version its more complicated.
I already extended the MicroSD-Card, so it is accessible from the outside of the PowerBook, and i really don't want also extend the MicroUSB :)
[image]

erichelgeson, 2023-10-07:
What version are you upgrading from? A few of the older releases did have an issue. I just tested 2023.08.10 to the latest snapshot and it worked. As someone who has opened up their powerbook 100's of times to do work on it I completely understand the frustration! If you could get on to the latest release via USB method I believe you should be fine going forward.

explit7, 2023-10-08:
I am trying to update from WifiBeta Firmware to the BlueSCSI Toolbox Pre-Release.
Or Updating to/from Beta Versions via USB is not supported/tested?
Thanks

erichelgeson, 2023-10-08:
I believe another user reported an issue when trying to update from the wifi beta to another release via SD card. Betas can have bugs and issues, why they're beta :) - I'll make a note to test this later on the wifi beta.

explit7, 2023-10-08:
thanks, undestood

fdanapfel, 2023-11-18:
Just tried updating the firmware on a BlueSCSIv2 DB25 (small board version) from the official [v2023.10.12](https://github.com/BlueSCSI/BlueSCSI-v2/releases/tag/v2023.10.12) to the official [v2023.11.16](https://github.com/BlueSCSI/BlueSCSI-v2/releases/tag/v2023.11.16) using the SD Card method and it didn't work. The [BlueSCSI_Pico_2023-11-17_bb06470b.bin](https://github.com/BlueSCSI/BlueSCSI-v2/releases/download/v2023.11.16/BlueSCSI_Pico_2023-11-17_bb06470b.bin) on the SD card was not detected as a new firmware and stayed on the card even after power cycling the BlueSCSIv2 multiple times.

So this issue doesn't seem to be specific to the wifi beta.

Updating using the USB method worked.

explit7, 2023-11-18:
thanks for the info @fdanapfel 
i was also pretty sure, that this bug has nothing to do with beta firmware.
the problem is somewhere else.
Firmware file is just ignored, doesn't matter, which fw version is running on the BlueSCSIv2.

explit7, 2024-02-23:
Thanks for the Firmware Update, but on PowerBook BlueSCSIv2 Model this bug is really annoying...

erichelgeson, 2024-02-23:
Unfortunately the upstream bootloader has quite a big issue that instead of continually trying to work around we need to fix. I know it's annoying but please bear with us - will be better in the end :)

explit7, 2024-02-23:
> Unfortunately the upstream bootloader has quite a big issue that instead of continually trying to work around we need to fix. I know it's annoying but please bear with us - will be better in the end :)

Thanks, I am happy, there is some work in progress!

explit7, 2024-04-03:
Hi @erichelgeson Is there some progress, for we can hope, this bug can be fixed in the next Firmware Update versions ? I mean, that has worked for you in the past, so that must be practically possible to update firmware from the SD card.

Old Powerbooks are very sensible machines. Plasticks broke. Every new Firmware Update, for which the whole PowerBook need to be dismantled is an risky act.

Please! 
Bring the next Update out, when this Problem is fixed! I need to wait as long, as it takes, but every Update in the manual way - is not an option for us PowerBook users...

Thanks

explit7, 2025-04-09:
Since there is still no progress on this issue, i am thinking what can be done, to solve the problem. Updating via USB-Method is NO GO for PowerBook Users like me. Every time disassembling the Powerbook to install Firmware Update will break the old hardware. We really need a solution here.
For the ZuluSCSI ther Autor writes on their [Github page](https://github.com/ZuluSCSI/ZuluSCSI-firmware) that there exist custom bootloader which can install firmware files from SD-Card. Could it be implemented in BlueSCSIv2?
What do you think @erichelgeson ?

It there is some hope, that it could be implemented?

The other solution for me would be switch to SCSIknife which is based on ZuluSCSI.

erichelgeson, 2025-04-09:
Well good news, we've made a lot of progress on getting the bootloader to work, I cant say it'll be in next release, but the release after. There is quite a bit of testing needing to be done with the approach and we'll need free time to do it. Sorry it's taken so long, but light at the end of the tunnel.

explit7, 2025-04-09:
Thanks for the Info, thats great!

explit28, 2025-07-08:
Many Thanks @erichelgeson ! It was worth the wait!