BlueSCSI/BlueSCSI-v2 #82: How to use Initiator mode?

fdanapfel, 2023-10-13:
The release notes for the latest firmware (https://github.com/BlueSCSI/BlueSCSI-v2/releases/tag/v2023.10.12) mention that "Initiator mode" is now supported, but there is no clear documentation on how to actually use this feature.

On the linked page (https://androda.work/bluescsi-initiator-mode-and-stock-update/) it mentions that two jumpers have to be set, but it doesn't mention anything about which jumpers those are.

Since that page also mentions "The first production versions of Desktop BlueSCSI which support Initiator mode are in manufacturing now.", does that mean that this feature will only work with new versions ofthe BlueSCSIv2 Desktop boards which will be available sometime in the future? Or will it also be possible to use it with the already existing BlueSCSIv2 boards?

Would be great if some information could be added in the Wiki (https://github.com/BlueSCSI/BlueSCSI-v2/wiki) describing the requirements and how to use this feature.

androda, 2023-10-13:
Hello,

The Initiator Mode feature is still very much "hot off the presses".  A wiki article and hardware release are in progress and will be completed soon.

fdanapfel, 2023-10-13:
@androda Thanks a lot for the quick response!

I see the Wiki has now been updated: https://github.com/BlueSCSI/BlueSCSI-v2/wiki/Initiator-Mode

So I guess I just have to wait for the new hardware version to be able to try this out.