BlueSCSI/BlueSCSI-v2 #83: Network no longer works after updating from WiFi Beta to new firmware released on 2023-10-12

fdanapfel, 2023-10-13:
After upgrading to firmware v2023-10-12 (https://github.com/BlueSCSI/BlueSCSI-v2/releases/tag/v2023.10.12) using the USB method, it is no longer possible to ping any other machines on the network from my Atari TT030.

Besides updating the firmware nothing else has changed in the setup, so I'm guessing one of the firmware changes that was introduced between the release of the Wifi beta and v2023-10-12 broke something that prevents the Wifi setup on the BlueSCSIv2 from working correctly.

Looking at the log.txt I've noticed that with firmware v2023-10-12 the entry showing that the connection to the WiFi network was successful is missing (with the beta it prints "Successfully connected to Wi-Fi SSID... " after the information about the voltage):
```
[1000ms] === Network Initialization ===
[1000ms] Unique board id: e6 61 38 52  83 81 86 35
[1863ms] Wi-Fi MAC: 00:80:19:52:83:81
[1863ms] Connecting to Wi-Fi SSID "<SSID>" with WPA/WPA2 PSK
[1880ms]  
[1880ms] Initialization complete!
[1895ms] INFO: Pico Voltage: 3.243V.
[21860ms] DBG BUS RESET
```

The bluescsi.ini only contains the settings required to enable WiFi Access (I've also added "Debug=1" as requested in the Troubleshooting section of the Wiki to get the debug output in the log.txt):
```
[SCSI]
Debug=1
WiFiSSID=<SSID>
WiFiPassword=<password>
# Optional - Each Pico-W will generate its own MAC address, so normally not needed.
# WiFiMACAddress=00:80:19:C0:FF:EE
```

I'm using the fully assembled BlueSCSIv2 DB25, Smaller Design from One Geek Army (https://shop.onegeekarmy.eu/index.php/product/bluescsi-v2-db25-smaller-design/) in case this makes any difference.

Attached are the log.txt from the working Wifi beta firmware, the log from the non working v2023-10-12  
[LOG_Wifi_Beta.txt](https://github.com/BlueSCSI/BlueSCSI-v2/files/12894174/LOG_Wifi_Beta.txt)
[LOG_2023-10-23.txt](https://github.com/BlueSCSI/BlueSCSI-v2/files/12894162/LOG_2023-10-23.txt)

mynameistroy, 2023-10-13:
What TCP stack are you using Stik or Mint?

I've had it working under Mint with no issues at all

fdanapfel, 2023-10-13:
I'm using Sting v1.26 on TOS 3.06 with the driver for Sting for the SCSILink from http://www.anodynesoftware.com/ethernet/main.html

mynameistroy, 2023-10-13:
I haven't tried Sting yet but I'll set it up later and see if I run into issues

mynameistroy, 2023-10-13:
I set up sting and have no issues that I can see.

Using Lichi to transfer files from my NAS right now with a problem.

fdanapfel, 2023-10-14:
Strange.

Do you use any additional settings in your bluescsi.ini besides the SSID and password for the network configuration? 

The reason I ask is because I found a post on a forum where somebody mentioned they had to adapt the "Vendor" and "Product" names to make it work: https://www.atari-forum.com/viewtopic.php?p=452237#p452237

fdanapfel, 2023-10-14:
After adding the following in the bluescsi.ini the network now works again on my Atari TT030 with firmware v2023.10.12:
```
[SCSI6]
Vendor = "Dayna   "
Product = "SCSI/Link       "
```

mynameistroy, 2023-10-14:
> Strange.
> 
> Do you use any additional settings in your bluescsi.ini besides the SSID and password for the network configuration?
> 
> The reason I ask is because I found a post on a forum where somebody mentioned they had to adapt the "Vendor" and "Product" names to make it work: https://www.atari-forum.com/viewtopic.php?p=452237#p452237
> 
> And also, which BlueSCSIv2 model are you using?
> 
> Edit: I now checked my WiFi-AP (Fritz!Box 7583), and when the Network on the TT isn't working with the current firmware the BlueSCSIv2 does not show up as a WiFi client. If I go back to the WiFi beta then then I can see a WiFi connection is established between the BlueSCSIv2 and the WIFi AP (Fritz!Box 7583).

Haha so that post was me and that was with the beta actually. The defaults were changed in the release firmware so that you didn't need to make the change for the Vendor/Product. It's odd that it is still troubling you in that fashion.

My bluescsi.ini is generally set to the manual defaults. What SCSI driver are you using?

fdanapfel, 2023-10-14:
Yes, it is strange, because with the beta firmware the emulated SCSILink was detected and working without having these two lines in the ini. I think the change of the Vendor/Product name was already in the last Wifi beta firmware.

If the names are not correct the Sting control panel to configure the network adapter will actually crash with 2 Bombs (found that out when I originally copied the lines from your forum post in the bluescsi.ini, since there is a typo in the product name (duplicate double-quotes in the beginning) you mentioned there).

I'm currently using HDDriver 11.14 as the SCSI driver on all my Ataris.

mynameistroy, 2023-10-14:
When you boot up without the product/vendor set in the ini what does HDDRIVER say the product/vendor is for the network device?

fdanapfel, 2023-10-14:
It says
`Atari SCSI 6.0: Dayna   SCSI/Link`

And now Sting also works without having the vendor/product names in the ini. Weird!

mynameistroy, 2023-10-14:
Very but I suppose it's working now which is the good thing.

mynameistroy, 2023-10-15:
Well if it's working now then I'll close this and we can revisit it if things go south again.

closed 2023-10-15 by mynameistroy