BlueSCSI/BlueSCSI-v2 #88: Issues installing OS X 10.2 on G4 Upgraded Power Macintosh 7600

evanboonie, 2023-10-24:
I have a Power Macintosh 7600 with an 800MHz G4 CPU upgrade. When attempting to install OS X 10.2 using XPostfacto, the machine will not boot. It works perfectly fine with the stock 200MHz 604e CPU and will also work with a 400MHz G3 upgrade I have if I set the CPU throttling option in XPosfacto to "8". No level of throttling seems to let the machine boot with the 800MHz CPU. When booting I just see an OpenFirmware screen with the message "Can't LOAD from this device". It loops on this forever, with the light on the BlueSCSI turning solid blue for several seconds, then going off again. The debug logs show "scsi_accel_rp2040_finishRead timeout" with some repeating commands. I have tried using the fastest SD card I have (Samsung PRO Plus) freshly formatted with the recommended SD formatting tool as suggested by the help docs, but this didn't change the behavior. I have attached a full log of letting it attempt to boot for a few cycles.
[log.txt](https://github.com/BlueSCSI/BlueSCSI-v2/files/13111492/log.txt)

androda, 2023-10-24:
Just to restate and make sure we're on the same page:
* Your BlueSCSI will boot the 7600 without issues, but only when using the 200MHz 604e or 400MHz G3 CPU
* When trying to use the G4 800, your machine no longer boots

Is this correct?

evanboonie, 2023-10-24:
That is correct in the case when trying to boot OS X 10.2. I am able to boot Mac OS 8.6 and 9.2.2 from the BlueSCSI without issue using any of the CPUs.

emellisa, 2024-01-08:
I have a small update on this. I recently acquired a 450MHz G4 CPU that will boot OS X 10.2, so the issue doesn't seem to be specific to a G4 CPU. It feels like the probelm has something to do with the clockspeed, as the faster the CPU I use, the higher I have to set the "CPU throttling" option in XPostfacto to. This issue doesn't seem to occur with a SCSI HDD, so it does seem to have something to do with the BlueSCSI. I'm not knowledgeable enough about the boot process and SCSI bus in these machines, but my rough guess is that the faster CPUs are trying to initialize the SCSI device faster and are not waiting long enough for the BlueSCSI to respond. I have also tried the v2023.11.16 firmware but that made no difference.

erichelgeson, 2024-05-22:
Could you try 1) powering via external power, and 2) using a different scsi cable? I've used BlueSCSI on G4's and very fast SCSI cards without issue. We have seen when the actual scsi controller is overclocked it can mess with timing, but the BlueSCSI and the controller can stay in sync and negotiate up to 10mb/sec. If you havent please also checkout the Troubleshooting guide in the wiki.

erichelgeson, 2024-09-15:
Please try the latest release which hopefully resolves these odd issues https://github.com/BlueSCSI/BlueSCSI-v2/releases/tag/v2024.09.15

erichelgeson, 2024-10-15:
Please let us know if you have any success or failure with the latest firmware, going to close for now though.

closed 2024-10-15 by erichelgeson