BlueSCSI/BlueSCSI-v2 #90: AppleTalk + 2 access points can't see eachother on same network.

sideburn, 2023-10-31:
I have a Wi-Fi extender on my network and when I connect one machine to that access point and another to my main router AP the two devices cannot see eachother over AppleTalk. 

Once both are connected to the same SSID then file sharing works properly. 

Tried Timbuktu as well. Same result. 

Config example: 
  • IIcx on system 7.1 with MacTCP connected to "Living Room" SSID (Wi-Fi extender)

 • Powebook on system 7.2 with OT connected to "Main" SSID (main router)

Devices cannot see eachother. Switch one to the same SSID and issue is resolved.

erichelgeson, 2023-11-02:
Apple Talk is notorious for not being route able over different networks. Do non-appletalk packets get routed? eg can you use MacPing to ping the a device on the other network? It is likely a limitation that the appletalk packets are not being routed to another network/subnet.

sideburn, 2023-11-02:
Ahh and I think Timbuktu is also using AppleTalk. I could try MacPing but I'm not sure how to set it up so I could ping the other machine.

erichelgeson, 2023-11-04:
Just ping any machine's IP address on the other network to ensure you can route to it.

erichelgeson, 2024-05-22:
It's been a while, if you have any other questions feel free to open a discussion to continue troubleshooting or if you have any questions.

closed 2024-05-22 by erichelgeson