BlueSCSI/BlueSCSI-v2 #93: Daynaport SCSI link connection doesn't re-connect after wake from sleep.

sideburn, 2023-11-04:
I just tested out Androda's 2023 10a PowerBook hardware with the wake from sleep fix and it looks like it has resolved the HD connection issue. I was able to remove my 5v power jumper wire bandaid. 

Sleep/wake and the drive "spins up" and resumes within seconds now. No more freeze for several minutes. 👍

The only issue now is resuming the tcp/ip connection. 

Upon wake up of you try to connect to something (telnet/Netscape etc) the system will freeze for a few minutes and then show an error that the connection has broken. 

Using the Wi-Fi D/A I can re-establish the Wi-Fi connection and I can change between access points but telnet, Netscape etc connection is broken so it looks like it's also the tcp/ip to Daynaport scsi link.

With open transport I am able to turn tcp/ip off and back on again but have not been able to reliably recover the connection. 

If I wait for a timeout and then to a warm reboot the tcp connection re-established. No need to power down and back up.

renekint, 2023-11-19:
I have similar issues (Quadra660AV/system7.5.3 & 7.6.1). However, in my case there are a few differences:
- The Fi-Wi/DA accessory doesn'y work (it doesn't fin the Daynaport)
- Waking up the TCP/IP connection can be done by using some other software, e.g. JCS's gopher client. It will re-connect the stack. So there is something on in the application side of things as well. 
- My computer sometimes freezes during TCP/IP access. Might be related.

My 2c of course ;-)

erichelgeson, 2024-05-21:
We had quite a long discussion on sleep/wake https://tinkerdifferent.com/threads/bluescsi-on-macintosh-portable.2764/ (quite long, feel free to skip to the end)

The theory we ended up with is that really a real DaynaPort was not meant to drop power while in use. The driver can not recover gracefully from this. Your solution of keeping it alive while sleeping is the right solution (for now). We're working on some other networking emulation fixes and features that may address this (or may not).

closed 2024-05-21 by erichelgeson