BlueSCSI/BlueSCSI-v2 #98: HD1.rom overwrites firmware?

reukiodo, 2023-11-19:
At least as of 20231012, the ROM drive function seems to overwrite at least part of the firmware, causing the BlueSCSI to become non-functional until firmware is re-flashed through USB.

1. flash firmware to Pico over USB
2. use known-working HD1.hda as test (ensure booting is successful)
3. rename working HD1.hda to HD1.rom
4. power up Pico and wait for writing
5. ... no more function

It may be coincidence (or not) but it seems that the Pico goes into an extremely high-pitched whine during HD1.rom writing, and this high-pitched whine continues after any successive reboots.

fdanapfel, 2023-11-19:
@reukiodo How big is the HD1.hda that you are trying to use as the ROM drive? Because as mentioned at https://github.com/BlueSCSI/BlueSCSI-v2/wiki/ROM-Drive the ROM drive can have a maximum size of 1692kb.

Does it also break if you try to use the ROM drive image from the wiki page?

reukiodo, 2023-11-27:
This was starting from ROM_Macintosh_System7_Formatted.rom from https://mega.nz/folder/A6JgxSgC#TtGXqkmvvGmXP-AYqNYUnA which I added a small System to.
[ROM_MacOS_7.zip](https://github.com/BlueSCSI/BlueSCSI-v2/files/13472675/ROM_MacOS_7.zip)

But the same is with any HD1.rom (of the correct size). It happens when starting with the blank, with the 6, and with custom made.

reukiodo, 2023-11-27:
[ROM_MacOS_8.zip](https://github.com/BlueSCSI/BlueSCSI-v2/files/13472695/ROM_MacOS_8.zip)

Another for comparison.

erichelgeson, 2025-06-19:
Will be in the next release: https://github.com/BlueSCSI/BlueSCSI-v2/releases/tag/2025-June-PreRel

closed 2025-06-19 by erichelgeson

erichelgeson, 2025-07-08:
Now in v2025.07.07