BlueSCSI/BlueSCSI-v2 #101: WiFi not working on powerbook 520C

silvi89UD, 2023-12-17:
Hi good morning, I'm silvano from italy. I have an iBook 520C running system 7.5. The hard disk image on the sd card works but I can't activate the internet wifi connection. I ve been installed daynaport application, create an empty NE4.hda file, and a BlueSCSI.ini file wich contains wifi's name and password.I've been installed openTransport and set "Alternate Ethernet" and "Using DHCP Server". But if try to open an internet browser like icab, f.e, I can't navigate because there's no Internet connection. I put the log.txt file.
Someone can help me please?? thanks bye!!
Platform: BlueSCSI Pico
FW Version: 2023.11.16-release Nov 17 2023 03:26:27
Flash chip size: 2048 kB
 
=== SD Card Info ===
SD card detected, FAT32 volume size: 30436 MB
SD Name: SD32G, MID: 0x03, OID: 0x53 0x44
 
=== Global Config ===
Reading configuration from bluescsi.ini
Active configuration:
 
=== Finding images in / ===
== Opening /NE4.hda for ID: 4 LUN: 0
---- Configuring as network based on image name
---- Image ready
== Opening /HD10_512 PowerBook 520c 240MB.hda for ID: 1 LUN: 0
---- Image ready

=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- Activating ROM drive, SCSI id 0 size 1692 kB
---- WARNING: This image does not appear to be a valid Macintosh Device image. See: https://github.com/BlueSCSI/BlueSCSI-v2/wiki/Disk-Images
---- Activated ROM drive, SCSI id 0 size 1692 kB
 
=== Configured SCSI Devices ===
* ID: 0, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 1692kB
* ID: 1, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 245824kB
* ID: 4, Type: Network, Quirks: Apple
 
=== Network Initialization ===
Wi-Fi MAC: 00:80:19:07:E3:38
Connecting to Wi-Fi SSID "" with no authentication
Wi-Fi connection failed: -4
 
Initialization complete!
INFO: Pico Voltage: 3.276V.

fdanapfel, 2023-12-17:
Based on the following messages in your log it looks like the BlueSCSIv2 is not able to connect to your WiFi Network:
```
Connecting to Wi-Fi SSID "" with no authentication
Wi-Fi connection failed: -4
```
Did you remove the SSID when posting the log. or is it really just saying "" in your log.txt? If the SSID is realy empty in the log. then you need to make sure that the value you have entered after `WiFiSSID=` in your bluescsi.ini exactly matches the SSID of your WiFi network.

silvi89UD, 2023-12-17:
Hi! I’ve Just copied the .log text. Yes i’ve check  the ssd name after “=“ si correct….

silvi89UD, 2023-12-18:
Hi good morning. I've been try with another wifi router but nothing. same error. The ssd name and pass are both correct....

fdanapfel, 2023-12-18:
Could you post the `bluescsi.ini` with the WiFi password removed?

Also adding `Debug=1` in the `[SCSI]` section of the `bluescsi.ini` might help to get more information about what is going wrong here.

silvi89UD, 2023-12-18:
SCSI]
WiFiSSID=Barazzutti-WiFi
WiFiPassword=
# Optional - Each Pico-W will generate its own MAC address, so normally not needed.
# WiFiMACAddress=00:80:19:C0:FF:EE

silvi89UD, 2023-12-18:
[bluescsi.ini.zip](https://github.com/BlueSCSI/BlueSCSI-v2/files/13703461/bluescsi.ini.zip)

silvi89UD, 2023-12-18:
now I try to add the debug=1 command

silvi89UD, 2023-12-18:
Platform: BlueSCSI Pico
FW Version: 2023.11.16-release Nov 17 2023 03:26:27
Flash chip size: 2048 kB
 
=== SD Card Info ===
SD card detected, FAT32 volume size: 30436 MB
SD Name: SD32G, MID: 0x03, OID: 0x53 0x44
 
=== Global Config ===
Reading configuration from bluescsi.ini
Active configuration:
 
=== Finding images in / ===
== Opening /NE4.hda for ID: 4 LUN: 0
---- Configuring as network based on image name
---- Image ready
== Opening /HD10_512 PowerBook 520c 240MB.hda for ID: 1 LUN: 0
---- Image ready

=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- Activating ROM drive, SCSI id 0 size 1692 kB
---- WARNING: This image does not appear to be a valid Macintosh Device image. See: https://github.com/BlueSCSI/BlueSCSI-v2/wiki/Disk-Images
---- Activated ROM drive, SCSI id 0 size 1692 kB
 
=== Configured SCSI Devices ===
* ID: 0, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 1692kB
* ID: 1, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 245824kB
* ID: 4, Type: Network, Quirks: Apple
 
=== Network Initialization ===
Wi-Fi MAC: 00:80:19:07:E3:38
Connecting to Wi-Fi SSID "" with no authentication
Wi-Fi connection failed: -4
 
Initialization complete!
INFO: Pico Voltage: 3.273V.

this is the log file with debub=1

fdanapfel, 2023-12-18:
OK, the entry for `WiFiSSID=` looks good to me. therefore I'm not sure why the BlueSCSI is not able to connect.  

Does your WiFi use 2.4GHz or 5GHz? If it is using 5GHz then this might be the issue, because I think the Raspberry Pi Pivo W can only use 2.4GHz WiFi.

BTW. the log looks the same as before. Did you add the `Debug=1` as I've mentioned before, or did you use `debug=1`? If you used `debug=1` then please change it to `Debug=1`.

silvi89UD, 2023-12-18:
is use Debug=1
maybe is this....I think that is 5Ghz wifi.....I try in other way to see if it works with an 2,4Ghz wi fi

silvi89UD, 2023-12-18:
I've been check and is 2,4Ghz..

silvi89UD, 2023-12-18:
I've tried another time..and now I've this error on log text
Platform: BlueSCSI Pico
FW Version: 2023.11.16-release Nov 17 2023 03:26:27
Flash chip size: 2048 kB
 
=== SD Card Info ===
SD card detected, FAT32 volume size: 30436 MB
SD Name: SD32G, MID: 0x03, OID: 0x53 0x44
 
=== Global Config ===
Reading configuration from bluescsi.ini
Active configuration:
 
=== Finding images in / ===
== Opening /NE4.hda for ID: 4 LUN: 0
---- Configuring as network based on image name
---- Image ready
== Opening /HD10_512 PowerBook 520c 240MB.hda for ID: 1 LUN: 0
---- Image ready

=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- Activating ROM drive, SCSI id 0 size 1692 kB
---- WARNING: This image does not appear to be a valid Macintosh Device image. See: https://github.com/BlueSCSI/BlueSCSI-v2/wiki/Disk-Images
---- Activated ROM drive, SCSI id 0 size 1692 kB
 
=== Configured SCSI Devices ===
* ID: 0, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 1692kB
* ID: 1, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 245824kB
* ID: 4, Type: Network, Quirks: Apple
 
=== Network Initialization ===
Wi-Fi MAC: 00:80:19:07:E3:38
Connecting to Wi-Fi SSID "" with no authentication
Wi-Fi connection failed: -4
 
Initialization complete!
INFO: Pico Voltage: 3.284V.
cyw43_send_ethernet failed: -6
cyw43_send_ethernet failed: -6
cyw43_send_ethernet failed: -6
cyw43_send_ethernet failed: -6
cyw43_send_ethernet failed: -6
cyw43_send_ethernet failed: -6
cyw43_send_ethernet failed: -6
cyw43_send_ethernet failed: -6
cyw43_send_ethernet failed: -6
cyw43_send_ethernet failed: -6
cyw43_send_ethernet failed: -6
cyw43_send_ethernet failed: -6
cyw43_send_ethernet failed: -6
cyw43_send_ethernet failed: -6
cyw43_send_ethernet failed: -6
cyw43_send_ethernet failed: -6
cyw43_send_ethernet failed: -6

erichelgeson, 2024-05-21:
Very sorry for this old issue sitting out here so long - did you get any resolution? I actually had someone have a similar issue and (oddly) a full overwrite format with the SD Formatter tool (in the wiki under troubleshooting) resolved it for them.

silvi89UD, 2024-05-22:
Hi good morning! don't worry about. No, I haven't solved the problem yet....maybe can I recontact you later? actually I've some others works and I haven't time......
thank you so much!

fdanapfel, 2024-08-31:
@silvi89UD In case you are still trying to solve this problem: as somebody else mentioned in https://github.com/BlueSCSI/BlueSCSI-v2/issues/163 for them the WiFi stopped working when they added a ROM drive.

According to the logs you posted you seem to have a ROM drive enabled on your BlueSCSI:
```
=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- Activating ROM drive, SCSI id 0 size 1692 kB
---- WARNING: This image does not appear to be a valid Macintosh Device image. See: https://github.com/BlueSCSI/BlueSCSI-v2/wiki/Disk-Images
---- Activated ROM drive, SCSI id 0 size 1692 kB
```

So maybe you could try to deactivate the ROM drive on the BlueSCSI to get the WiFi connection working.

erichelgeson, 2024-09-15:
It's been quite a while on this issue and I've used wifi on my 540 (same motherboard, different processor board) without issue. If you still have an issue happy to help debug it, but closing for now.

closed 2024-09-15 by erichelgeson