BlueSCSI/BlueSCSI-v2 #114: Link to this .HDA image is not working

Stefan-van-Hattum, 2024-01-24:
In the manual under "Using MacTCP" 
1. Load [this .HDA image](https://mega.nz/file/4kpDkQAb#-EEw3HZRrHOgDmK9qKRrM-zrh9OIg55gXgQn6T_JCXM) to your BlueSCSI, it has the following software installers:

The link provided is giving an error "The file could not be loaded" 
Tried different browsers from different computers.

erichelgeson, 2024-01-24:
I just checked the link on bluescsi.com and on the wiki - both work fine. The link in this issue also is fine. Could you provide more details on what error you're getting?

Stefan-van-Hattum, 2024-01-24:
Ok thank you for confirming the link is ok.
I just tried on my game PC and installed the Mega Desktop application and I am able to download the image now.

erichelgeson, 2024-01-24:
Glad you got it working, though I've never used their desktop app. Works fine on windows/mac/linux (i just tested) - firefox and chrome.

closed 2024-01-24 by erichelgeson