BlueSCSI/BlueSCSI-v2 #115: Mess regarding versioning of printed brackets and cases

plaes, 2024-01-26:
Currently, with new hardware revisions, older bracket and case files are overwritten in the repository, which makes it quite a feat to track down a compatible 3D-printed file for specific hardware revision in case there have been any changes in fastener layouts.

It would be a lot nicer if these STL files have some kind of revision info in their filename and optionally with a README including which revision of hardware is meant for which bracket/case.

(This was written after printing out two incompatible brackets for 50pin desktop variant rev. 2023.03a for which there apparently is not any compatible bracket available in this repo.)

erichelgeson, 2024-01-28:
Some things we can do:

* Tag each hardware release so you can more easily find the brackets in past commits. 
* Break up the wiki page into current and past brackets 

Also we don't foresee having to change the mounting holes again, we just needed to for routing and better placement of the usb.

Thanks for the report on the BlueSCSI working on that workstation - if you don't mind you could add that to our compatibility page.

erichelgeson, 2024-03-30:
I've taken all the stock cases out and added them to Printables:

* https://www.printables.com/model/826134-bluescsi-v2-50pin-desktop-stock-case
* https://www.printables.com/model/826137-bluescsi-v2-powerbook-stock-case/files
* https://www.printables.com/model/826125-bluescsi-v2-centronics-stock-case-202303a
* https://www.printables.com/model/826131-bluescsi-v2-db25-stock-case

I've also created a BlueSCSI v2 collection with a lot of other cool community ones: https://www.printables.com/@EricHelgeson_1492862/collections/1024938

Lastly I've updated the documentation as well here https://github.com/BlueSCSI/BlueSCSI-v2/wiki/Accessories#3d-cases

Thanks again for reporting it - sorry it took a while for me to get it done.

closed 2024-03-30 by erichelgeson