BlueSCSI/BlueSCSI-v2 #158: Q280 doesn't seem to image in initiator mode

Kirkman, 2024-07-02:
I'm probably doing something wrong, but my first attempts to image a Q280 80-meg hard drive with a Blue SCSI have not worked.

It *looked* and *sounded* like it was working. I powered up the hard drive, and its LED came on. Then I powered up the BlueSCSI, and its LED flashed then became solid. The hard drive spun up and it sounded like it was busy reading data. This went on for a long time, longer than I would have thought. 

I decided to check the log file on the SD card to see if everything was working okay. I powered down the drive, then powered down the Blue SCSI and checked the SD card.

There was no .HD image on the card. There was only the .INI and log. The log looked like this:

```
Platform: BlueSCSI Pico
FW Version: 2023.11.17-dev Feb 12 2024 21:57:14
I2C Supported
Flash chip size: 2048 kB
SCSI Initiator Mode.  Will scan the bus for drives to image.
 
=== SD Card Info ===
SD card detected, exFAT volume size: 121910 MB
SD Name: SK128, MID: 0x03, OID: 0x53 0x44
InitiatorID set to ID 7
 
Initialization complete!
INFO: Pico Voltage: 3.284V.
RequestSense response: 0x70 0x00 0x02 0x00 0x00 0x00 0x00 0x0A 0x00 0x00 0x00 0x00 0xB0 0x00 0x00 0x00 0x00 0x00 
Target 0 reports NOT_READY, running STARTSTOPUNIT
RequestSense response: 0x70 0x00 0x06 0x00 0x00 0x01 0x00 0x0A 0x00 0x00 0x00 0x00 0x29 0x00 0x00 0x00 0x00 0x00 
START STOP UNIT on target 0 failed, sense key 0x06
RequestSense response: 0x70 0x00 0x02 0x00 0x00 0x00 0x00 0x0A 0x00 0x00 0x00 0x00 0xB0 0x00 0x00 0x00 0x00 0x00 
Target 0 reports NOT_READY, running STARTSTOPUNIT
* No response from SCSI ID 0
* No response from SCSI ID 1
* No response from SCSI ID 2
* No response from SCSI ID 3
* No response from SCSI ID 4
```

I tried again with debugging turned on. This time, I powered the BlueSCSI first, and then the drive. But the pattern was the same ... BlueSCSI reports trying to communicated with SCSI 0 for about 1 second. Then no further log entries for 8 minutes, at which point I powered off the hard drive. Then it resumed scanning the bus.

I'll post the second log below. Any ideas what I'm doing wrong?

```
Platform: BlueSCSI Pico
FW Version: 2023.11.17-dev Feb 12 2024 21:57:14
I2C Supported
Flash chip size: 2048 kB
SCSI Initiator Mode.  Will scan the bus for drives to image.
 
=== SD Card Info ===
SD card detected, exFAT volume size: 121910 MB
SD Name: SK128, MID: 0x03, OID: 0x53 0x44
[505ms] InitiatorID set to ID 7
[1713ms]  
[1713ms] Initialization complete!
[1748ms] INFO: Pico Voltage: 3.295V.
[3029ms] DBG -- BUS_BUSY
[3029ms] DBG ---- SELECTION
[3029ms] DBG ------ SELECTING 0 with initiator ID 7
[3997ms] DBG ------ Target 0 did not respond
[3997ms] DBG -- BUS_FREE
[4000ms] * No response from SCSI ID 0
[5012ms] DBG -- BUS_BUSY
[5012ms] DBG ---- SELECTION
[5012ms] DBG ------ SELECTING 1 with initiator ID 7
[5264ms] DBG ------ Target 1 did not respond
[5264ms] DBG -- BUS_FREE
[5267ms] * No response from SCSI ID 1
[6282ms] DBG -- BUS_BUSY
[6282ms] DBG ---- SELECTION
[6282ms] DBG ------ SELECTING 2 with initiator ID 7
[6534ms] DBG ------ Target 2 did not respond
[6534ms] DBG -- BUS_FREE
[... 212 lines cut ...]
[603762ms] DBG -- BUS_BUSY
[603762ms] DBG ---- SELECTION
[603762ms] DBG ------ SELECTING 4 with initiator ID 7
[604014ms] DBG ------ Target 4 did not respond
[604014ms] DBG -- BUS_FREE
[604017ms] * No response from SCSI ID 4
[605029ms] DBG -- BUS_BUSY
[605029ms] DBG ---- SELECTION
[605029ms] DBG ------ SELECTING 5 with initiator ID 7
[605281ms] DBG ------ Target 5 did not respond
[605281ms] DBG -- BUS_FREE
[605284ms] * No response from SCSI ID 5
[606294ms] DBG -- BUS_BUSY
[606294ms] DBG ---- SELECTION
[606294ms] DBG ------ SELECTING 6 with initiator ID 7
[606546ms] DBG ------ Target 6 did not respond
[606546ms] DBG -- BUS_FREE
[606549ms] * No response from SCSI ID 6
[607561ms] DBG -- BUS_BUSY
[607561ms] DBG ---- SELECTION
[607561ms] DBG ------ SELECTING 0 with initiator ID 7
[607814ms] DBG ------ Target 0 did not respond
[607814ms] DBG -- BUS_FREE
[607816ms] * No response from SCSI ID 0
```

androda, 2024-07-02:
According to [the drive manual](http://www.bitsavers.org/pdf/quantum/Q200/81-45416_Q200_Series_Programmers_Manual_Nov87.pdf) it looks like these codes indicate the drive is having a hard time spinning up and stabilizing at operating speed.

In the first log messages, the logs say:
* Sense Code 0x02 is 'not ready' and additional code '0xB0' indicates the drive is in recalibration.
* Sense code 0x06 is 'unit attention' and additional code '0x29' indicates the drive entered a reset condition.
* Sense Code 0x02 is 'not ready' and additional code '0xB0' indicates the drive is in recalibration.
No response to commands following this.

In the second set of log messages, we have this series of events:
* Sense Code 0x02 is 'not ready' and additional code '0xB1' indicates the drive has not reached a stable working RPM for at least two seconds.
* Sense code 0x06 is 'unit attention' and additional code '0x29' indicates the drive entered a reset condition.
* Sense Code 0x02 is 'not ready' and additional code '0xB1' indicates the drive has not reached a stable working RPM for at least two seconds.

So the drive succeeded in performing recalibration (probably does this regularly every certain number of spin-ups) because we no longer see B0 in the second set of log messages.  But B1 isn't any better, the drive can't reach a stable RPM and then fails to respond to further commands which is triggering the watchdog reset.

Kirkman, 2024-07-02:
Thanks so much. It sounds like this is a drive issue rather than Blue SCSI, then. Can you recommend any resources where I could find a solution to these drive issues? (I'm wondering if it can't get stable RPM, should I try lubricating something? Switch out the power supply?  etc)