BlueSCSI/BlueSCSI-v2 #173: SCSI_NETWORK_WIFI_CMD_SCAN_RESULTS ignores transfer size

th-otto, 2024-07-24:
In https://github.com/BlueSCSI/BlueSCSI-v2/blob/df274ea5aae5b7b2cd57a24ffafe9c981f4490a5/lib/SCSI2SD/src/firmware/network.c#L328, the size of the returned data is only checked against the size of the internal data buffer. But that ignores the size of the initiators specified transfer size in bytes 3/4 of the CDB. 

I don't know if this can cause trouble, since at most 10 wifi_network_entries are returned (using 2 + 10 * 74 bytes), but IMHO this should be fixed.

BTW, is there any way to increase that number of entries (WIFI_NETWORK_LIST_ENTRY_COUNT)?

th-otto, 2024-07-24:
I understand the limitation, but the problem is that even "at home", their may be entries from your neighbours that you are not interested in (unless you know their passwords ;)

I guess its too late to change the interface, so the burden of storing/filtering the entries can be put to the host application instead?

erichelgeson, 2024-07-24:
Just brainstorming.. since `SCSI_NETWORK_WIFI_CMD_SCAN_RESULTS` is the only place that `wifi_network_list` is used - instead of storing it in the `wifi_network_list` we can write entries directly to `scsiDev.data` (as is done in many places.. for better or worse) to save memory. This could be done in a compatible/non-breaking change to the existing mac wifi-da.

jcs, 2024-07-24:
I think the intention was to sort by RSSI so that closer/stronger networks would appear first in the list, making it less likely that the one you care about is far down the list.  But I guess I never wrote that.

https://github.com/BlueSCSI/BlueSCSI-v2/blob/df274ea5aae5b7b2cd57a24ffafe9c981f4490a5/lib/BlueSCSI_platform_RP2040/BlueSCSI_platform_network.cpp#L173

th-otto, 2024-07-24:
> instead of storing it in the `wifi_network_list` we can write entries directly to `scsiDev.data`

But is that safe? What if some other command is sent, while the scan is in progress (possibly from some other application).

erichelgeson, 2024-07-24:
>But is that safe? What if some other command is sent, while the scan is in progress (possibly from some other application).

During this phase the BlueSCSI is selected and has exclusive control over the bus, it should be safe.

th-otto, 2024-07-24:
PR just opened. I've no way to test it, but checked that it compiles atleast.

erichelgeson, 2024-09-22:
cherry-picked and merged.

closed 2024-09-22 by erichelgeson