BlueSCSI/BlueSCSI-v2 #175: after HD[X]9_imaged.hda is written the next file isn't writable

dericed, 2024-08-21:
On my SD card, I have these files: 
```
HD60_imaged.hda
HD61_imaged.hda
HD62_imaged.hda
HD63_imaged.hda
HD64_imaged.hda
HD65_imaged.hda
HD66_imaged.hda
HD67_imaged.hda
HD68_imaged.hda
HD69_imaged.hda
bluescsi.ini
log.txt
```

When I begin a subsequent transfer, I get these lines in the log:

```
[22595ms] Using filename: HD6:_imaged.hda to avoid overwriting existing file.
[22595ms] Failed to open file for writing: HD6:_imaged.hda
```

Perhaps the increment sets a `:` (colon) as the value after `9`. 

I can work around this by removing the existing files.

erichelgeson, 2024-08-21:
Thanks for reporting! Yes it's due to me just simply incrementing the lun, which when it gets to 9 in ascii the next value is `:`.  A simple printf or something would be better instead of modifying that one byte. https://github.com/BlueSCSI/BlueSCSI-v2/blob/main/src/BlueSCSI_initiator.cpp#L271 

A good first issue for someone who might want to contribute.

closed 2024-09-22 by erichelgeson

dericed, 2024-09-30:
thank you! :D