BlueSCSI/BlueSCSI-v2 #210: Retry for sdio write errors.

nsloss, 2024-11-20:
Hi,

I've updated to the latest 2024-10-26 release
And adding clock delay 
[SDIO]
AddClockDelay=1

Still resulted in errors in the log.txt

Platform: BlueSCSI
FW Version: 2024.10.26-rel Oct 26 2024 22:02:38
I2C Supported
Flash chip size: 2048 kB
INFO: Injecting 1 additional wait state(s) on SDIO
 
=== SD Card Info ===
SD card detected, exFAT volume size: 61088 MB
SD Name: EC1S5, MID: 0x1B, OID: 0x53 0x4D
 
=== Global Config ===
Reading configuration from bluescsi.ini
Active configuration:
 
=== Finding images in / ===
== Opening /HD00_512.hda for ID: 0 LUN: 0
---- Image ready
== Opening /NE4.hda for ID: 4 LUN: 0
---- Configuring as network based on image name
---- Image ready

=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- ROM drive image not detected
 
=== Configured SCSI Devices ===
* ID: 0, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 20480000kB
* ID: 4, Type: Network, Quirks: Apple
 
=== Network Initialization ===
Wi-Fi MAC: xx:xx:xx:xx:xx:xx
Connecting to Wi-Fi SSID "MYSSID" with WPA/WPA2 PSK
 
Initialization complete!
INFO: Pico Voltage: 3.357V.
Successfully connected to Wi-Fi SSID "NATMAN"
SDIO card reports write CRC error, status 0x000000EB
SdioCard::writeSectors(0x0075F2A0,...,16) failed: 7
SD card write failed: 0x00

Retrying on SDIO WRITE ERRORS was successful after applying the following patch:
--- ./lib/BlueSCSI_platform_RP2040/rp2040_sdio.cpp      2024-10-27 05:41:00.000000000 +1100
+++ /home/nat/rp2040_sdio.cpp   2024-11-20 18:58:31.423144365 +1100
@@ -56,6 +56,7 @@ static struct {
     uint32_t end_token_buf[3]; // CRC and end token for write block
     sdio_status_t wr_status;
     uint32_t card_response;
+    uint8_t retries;
 
     // Variables for block reads
     // This is used to perform DMA into data buffers and checksum buffers separately.
@@ -706,6 +707,8 @@ sdio_status_t rp2040_sdio_tx_start(const
     g_sdio.total_blocks = num_blocks;
     g_sdio.blocks_checksumed = 0;
     g_sdio.checksum_errors = 0;
+    g_sdio.retries = 0;
+    g_sdio.wr_status = SDIO_OK;
 
     // Compute first block checksum
     sdio_compute_next_tx_checksum();
@@ -713,12 +716,6 @@ sdio_status_t rp2040_sdio_tx_start(const
     // Start first DMA transfer and PIO
     sdio_start_next_block_tx();
 
-    if (g_sdio.blocks_checksumed < g_sdio.total_blocks)
-    {
-        // Precompute second block checksum
-        sdio_compute_next_tx_checksum();
-    }
-
     return SDIO_OK;
 }
 
@@ -788,10 +785,25 @@ static void rp2040_sdio_tx_irq()
             g_sdio.wr_status = check_sdio_write_response(g_sdio.card_response);
 
             if (g_sdio.wr_status != SDIO_OK)
-            {
-                rp2040_sdio_stop();
-                return;
-            }
+           {
+               g_sdio.retries++;
+               if (g_sdio.retries > 3)
+               {
+                    rp2040_sdio_stop();
+                    return;
+               }
+               else if (g_sdio.blocks_done < g_sdio.total_blocks)
+               {
+                   g_sdio.wr_status = SDIO_OK;
+                    sdio_start_next_block_tx();
+                    g_sdio.transfer_state = SDIO_TX;
+                   return;
+               }
+           }
+
+           g_sdio.retries = 0;
+            if (g_sdio.blocks_checksumed < g_sdio.total_blocks)
+                sdio_compute_next_tx_checksum();
 
             g_sdio.blocks_done++;
             if (g_sdio.blocks_done < g_sdio.total_blocks)
@@ -799,12 +811,6 @@ static void rp2040_sdio_tx_irq()
                 sdio_start_next_block_tx();
                 g_sdio.transfer_state = SDIO_TX;
 
-                if (g_sdio.blocks_checksumed < g_sdio.total_blocks)
-                {
-                    // Precompute the CRC for next block so that it is ready when
-                    // we want to send it.
-                    sdio_compute_next_tx_checksum();
-                }
             }
             else
             {

This will retry writing the current block up to 3 times and so far this has been successful reporting no errors on NetBSD-current

BlueSCSI-v2 with wifi Powebook Edition.
20G Image size.

I hope this helps,

Best regards,

Nat

nsloss, 2024-11-20:
reopening...  my apologies im a beginner

erichelgeson, 2024-11-20:
Thanks for taking the time to dig in and no need to apologize! 

I see you tested with the 10/26 release - Is there anyway you could re-test with the [latest nightly](https://github.com/BlueSCSI/BlueSCSI-v2/releases/tag/latest) (without your patch) and see if you experience the same result? We just merged #209 which _should_ help with this case, but it'd be great to know if it did/didnt.

Also you say you're running NetBSD on your powerbook? Cool! I thought the NetBSD driver wasn't compatible or was there an update? Either way any updates to the wiki with any tips or tricks (or brebuilt base image) would be appreciated.

closed 2024-12-08 by erichelgeson