BlueSCSI/BlueSCSI-v2 #212: Implement a USB->SCSI Bridge Mode for BlueSCSI

akuker, 2024-11-20:
[image]

This feature request is to add a new "mode" to BlueSCSI that allows it to connect legacy SCSI devices to a modern-ish host via USB. The goal is to emulate a device similar to this:
[image]

The scope of this issue will be to add basic mass storage support (hard drive, cd-rom) to BlueSCSI on a Linux host. (Linux seems to be the least picky about USB device drivers). Initial implementation will keep the CDC (console) port over USB to allow debugging. MacOS 9 might not handle composite USB devices properly, so we may need an alternate config for that.

Once this basic functionality is implemented, the following wish-list will be evaluated:
- MacOS 8/9 support
- Modern Windows (11) support
- Modern macOS (13) support
- [SMDI Support (for samplers)](https://www.exellon.net/book/smdi.pdf)
- Scanner support - Potentially integrate with [VueScan](https://www.hamrick.com/) or other modern software
- Emulate specific devices potentially including:
  - [Adaptec USB2Xchange](https://download.adaptec.com/pdfs/user_guides/usb2xchange_v10_ug.pdf), 
  - [Ratoc U2SCX](https://www.ratocsystems.com/english/download/brochure/u2scx.pdf) 
  - Castlewood ORB SCSI to USB adapter
  - [Microtech XpressSCSI USB to SCSI adapter](https://www.macintoshrepository.org/51518-microtech-usb-xpressscsi)
  - [Belkin USB to SCSI Adapter](https://www.macintoshrepository.org/39688-belkin-usb-scsi-adapter-with-termpower-f5u015-)  
  - Shuttle eUSCSI Bridge
- Tape drive support

References:
- [USB Mass Storage overview](https://usb.org/sites/default/files/Mass_Storage_Specification_Overview_v1.4_2-19-2010.pdf)
- PiSCSI issue requesting same function: https://github.com/PiSCSI/piscsi/issues/496

ewanuno, 2024-11-20:
https://www.exellon.net/book/smdi.pdf

this is the smdi specification for transafering audio to and from samplers.

ewanuno, 2024-11-20:
https://macintoshrepository.org/182-akai-mesa-ii-
this is Akai Mesa II which runs on os9 and connects to Akai xl series samplers
there is also a windows version.

akuker, 2024-11-20:
Castlewood Orb USB Smart Cable info:
```
Bus 001 Device 012: ID 084b:a001 Castlewood Systems Castlewood ORB Drive
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        16
  idVendor           0x084b
  idProduct          0xa001
  bcdDevice            2.06
  iManufacturer           1 Castlewood Systems
  iProduct                2 Castlewood ORB Drive
  iSerial                 5 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          3 Mass Storage Class
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              400mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              4 Bulk-Only and CBC Interface
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

akuker, 2024-11-20:
Microtech SCSI-HD50 SCSI Bridge
```

Bus 001 Device 013: ID 07af:0005 Microtech SCSI-HD50 SCSI Bridge [shuttle]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x07af Microtech
  idProduct          0x0005 SCSI-HD50 SCSI Bridge [shuttle]
  bcdDevice            1.00
  iManufacturer           1 Microtech International, Inc.
  iProduct                2 USB-SCSI-HD50
  iSerial                 5 07
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          3 Default
    bmAttributes         0x40
      (Missing must-be-set bit!)
      Self Powered
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              4 Mass storage
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
cannot read device status, Resource temporarily unavailable (11)
```

ewanuno, 2024-11-21:
this post on stackexchange has some info about adapters supported by linux:
https://superuser.com/questions/773331/attaching-scsi-drive-through-usb

ewanuno, 2024-11-21:
a user confirms that the castlewood orb scsi adapter works with their EMU sampler:
https://forum.vintagesynth.com/viewtopic.php?t=62242

ewanuno, 2024-11-21:
linux kernel patch with more product names for usb-scsi devices.
seems like lots of them use the shuttle technology eusb-01 chip
https://www.spinics.net/lists/stable/msg64565.html

ewanuno, 2024-11-21:
teardown of a shuttle eusb-01 based device over at thinkgeek:
https://tinkerdifferent.com/threads/scsi-to-usb-adapter-reverse-engineering.1878/

stinkerton18, 2024-11-21:
Adaptec Blue and White USB2-Xchange

Kernel message (no drive connected):
`[ 1373.355758] usb 3-1: new high-speed USB device number 8 using xhci_hcd
[ 1373.481955] usb 3-1: New USB device found, idVendor=03f3, idProduct=2002, bcdDevice= 1.00
[ 1373.481970] usb 3-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
`

LSUSB output:
`Bus 003 Device 008: ID 03f3:2002 Adaptec, Inc. USB2-Xchange
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol       255 Vendor Specific Protocol
  bMaxPacketSize0        64
  idVendor           0x03f3 Adaptec, Inc.
  idProduct          0x2002 USB2-Xchange
  bcdDevice            1.00
  iManufacturer           0
  iProduct                0
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00ab
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           6
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
[... 171 lines cut ...]
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x88  EP 8 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol       255 Vendor Specific Protocol
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)`

dramirez1972, 2024-11-21:
Entrega USB to SCSI Converter:

```
pi@raspberrypi4:~ $ lsusb -v

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         3
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            6.01
  iManufacturer           3 Linux 6.1.21-v8+ xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 0000:01:00.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        bMaxBurst               0

Bus 001 Device 003: ID 1645:0007 Entrega [hex] U1-SC25 SCSI
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1645 Entrega [hex]
  idProduct          0x0007 U1-SC25 SCSI
  bcdDevice            1.00
  iManufacturer           1 Entrega Technologies Inc.
  iProduct                2 Entrega USB to SCSI Converter
[... 117 lines cut ...]
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
pi@raspberrypi4:~ $
```

n9yty, 2024-12-11:
Wow - this would be incredible!!

I have been fighting to get this Entrega U1-SC25 working on ANYTHING - Windows XP in a VM, Linux, MacOS 9, Mac OS X, no luck...

```
Bus 002 Device 004: ID 1645:0007 Entrega [hex] U1-SC25 SCSI
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1645 Entrega [hex]
  idProduct          0x0007 U1-SC25 SCSI
  bcdDevice            1.00
  iManufacturer           1 Entrega Technologies Inc.
  iProduct                2 Entrega USB to SCSI Converter
  iSerial                 5 07
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          3 MSD Configuration
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              160mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              4 BONLY and CBC Interface
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
```

c-alpha, 2025-03-31:
I'm all for it! The bridge mode would be a huge thing. Two application areas:

- **Software/data Archeology.** For instance I have a **Atari System VR4 boot tape set**. Perhaps the only surviving copy on the planet? I also have old QIC drives ot read it, but no host computer.
- **Professional Grade Peripherals From the Scrap Yard.** Comments on this thread mention film scanners, and other cool stuff you can get new for the price of a house only.

If there were a pass-through mode of sorts supported in the hard- and firmware, how should the USB interface for it look like? An extended version of UAS? Some new invention?

In any case, the question remains: what software are we going to run on the modern computer to access the data coming from the SCSI device?

In the old days and on unixoid platforms, the kernel's SCSI driver would expose devices on the SCSI bus as device files (for example `/dev/rmt0` for the first tape device; "raw magnetic tape number zero"). Applications would then open those device files, and perform `read`, `write`, and `ioctl` operations ([example](https://docs.oracle.com/cd/E36784_01/html/E36884/mtio-7i.html#REFMAN7mtio-7i)) on them.

USB-IF have defined a couple of device class specifications, but not for all device types for which standardised command sets exist on SCSI. Hence, those missing command sets would need to be implemented on the modern computer. That sounds like a lot of work, since the whole list mentioned in @akuker's issue descriptiopn would need to be implemented on three platforms (Linux, Windows, macOS). Not to say it couldn't be done, but it sure looks like a **lot** of work.

erichelgeson, 2025-07-08:
Now in v2025.07.07. If there are other modes or updates needed please raise separate issues for those.

closed 2025-07-08 by erichelgeson

erichelgeson, 2025-09-02:
> Are there authorized vendors that sell external hardware with a USB port that works in bridge mode?

You will need a 50 pin model and an adapter to whatever connector you'd like. It needs initiator mode to bridge the connection, which the 50 pin has.

Right now it's bridging storage but as c-alpha states modern drivers could be made for scanners or other devices, though they wont work out of the box as there's no generic simple 1-1 mapping of how non-storage drivers communicate over scsi and over usb.