BlueSCSI/BlueSCSI-v2 #230: Watchdog timeouts when in initiator mode causing bus reset and invalid state.

me0262, 2025-01-26:
I'm attempting to use BlueSCSI in initiator mode to image a failing hard drive.
The drive appears to reset itself quite often when attempting to read from it, causing long delays when receiving information.

Logfile attached. Pertinent information is at line 1065 where the bus reset causes the device to go into an invalid state and will no longer attempt to read anything.

[HD0_imaged-004.log.txt](https://github.com/user-attachments/files/18548999/HD0_imaged-004.log.txt)

Here are a few things that could also be implemented in Initiator mode:
 - configurable watchdog timer
 - configurable number of retries before giving up
 - options to skip certain sectors (known bad and potentially problematic)
 - log file as image name (HD0_imaged-001.log.txt to match with HD0_imaged-001.hda)

erichelgeson, 2025-01-29:
Thanks for the report - defiantly looks like we could relax some timings for older, tired drives that can't keep up or have lots of bad sectors.

Skip known bad sectors - how would you know about this ahead of time? We have thought about logging bad sectors to a separate file and then allow only retries of bad sectors, but not to skip - expand on this a bit more please.

Are these Items you'd like to take a stab at?

me0262, 2025-02-18:
> Skip known bad sectors - how would you know about this ahead of time? We have thought about logging bad sectors to a separate file and then allow only retries of bad sectors, but not to skip - expand on this a bit more please.
This came from multiple runs trying to break through the watchdog. Maybe progress much like how `ddrescue` works that can stop and resume. Grab data, print bad sector list to a file, option to ignore/wipe bad sector file to go back over once good data is retrieved. "Recovery Mode"?

> Are these Items you'd like to take a stab at?
If I familiarize myself with more of the code base, probably, but I'm very busy on a lot of other personal projects (including restoring an SGI workstation that BlueSCSI helped me image the drive). I'll see, because it is a personal drive of mine that needs this slack.