BlueSCSI/BlueSCSI-v2 #258: ExFat gives incorrect size for HD images

brijohn, 2025-04-28:
I recently got a BlueSCSI for my old MS-DOS system and was having problems correctly reading HD images when formatting the SD Card as ExFAT (FAT32 worked fine).  Here is the log generated during initialization on an ExFAT system. ID 0 is the 512MB DOS image downloaded from the BlueSCSI site, while ID 1 is a zero filled 512MB image created using DD. The size for both files is clearly wrong.

Doing some digging this seems to be due to using the FsFile.size method from SDFat for retrieving the the files size, however when running on an ExFAT system instead of returning the actual allocated size (dataLength) of the file it returns the validDataLength field instead, which can be less then the actual number of bytes in a file.

This has been kind of fixed in the Upstream 2.2.2 version of ExFAT where a new method called dataLength was added do FsFile that returns either fileSize on a FAT system or dataLength on an ExFAT system. The actual size method is still broken though as it returns two different values depending on if the underlying filesystem is FAT or ExFAT.

```
Platform: BlueSCSI
FW Version: 2024.12.08-rel Dec  8 2024 22:19:03
Flash chip size: 2048 kB
 
=== SD Card Info ===
SD card detected, exFAT volume size: 119844 MB
SD Name: USD00, MID: 0xAD, OID: 0x4C 0x53
 
=== Global Config ===
Reading configuration from bluescsi.ini
Active configuration (using system preset "Generic"):
-- SCSI ID: 2 using Optical image directory 'CD2'
-- Opening 'CD2/Fallout 1.1.iso' for ID: 2
 
=== Finding images in / ===
== Opening /HD00_512_512MB_DOS622.img for ID: 0 LUN: 0
---- Image ready
== Opening /HD10_512_512MB_DOS_Games.img for ID: 1 LUN: 0
---- Error: image file /HD10_512_512MB_DOS_Games.img is empty
---- Failed to load image

=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- ROM drive image not detected
 
=== Configured SCSI Devices ===
* ID: 0, BlockSize: 512, Type: Fixed, Quirks: None, Size: 27392kB
* ID: 1, BlockSize: 512, Type: Fixed, Quirks: None, Size: 0kB
* ID: 2, BlockSize: 2048, Type: Optical, Quirks: None, Size: 618240kB
 
Initialization complete!
```

erichelgeson, 2025-04-28:
Thanks for the detailed report! I'll update our fork to 2.2.3 and push a build this evening I hope if you'd like to test.

erichelgeson, 2025-06-19:
Please test the latest pre-release - updated to 2.2.3 there. https://github.com/BlueSCSI/BlueSCSI-v2/releases/tag/2025-June-PreRel

brijohn, 2025-06-25:
Took me a bit to get around to running a test.

```
Platform: BlueSCSI
FW Version: 2025.06.00-dev Jun 19 2025 18:42:53
Flash chip size: 2048 kB
 
=== SD Card Info ===
SD card detected, exFAT volume size: 119844 MB
SD Name: USD00, MID: 0xAD, OID: 0x4C 0x53

=== Global Config ===
Reading configuration from bluescsi.ini
Using system preset "Generic")

=== Finding images ===
== Opening 'CD2/Fallout 1.1.iso' for ID 2
---- Configuring as CD-ROM drive
=== Finding images in / ===
== Opening /HD00_512_512MB_DOS622.img for ID:0 LUN:0
---- Configuring as disk drive
== Opening /HD10_512_512MB_DOS_Games.img for ID:1 LUN:0
---- Configuring as disk drive

== Platform supports ROM drive up to 1466 kB
 
=== Configured SCSI Devices ===
ID: 0, BlockSize: 512, Type: Fixed, Quirks: None, Size: 27392kB
ID: 1, BlockSize: 512, Type: Fixed, Quirks: None, Size: 524288kB
ID: 2, BlockSize: 2048, Type: Optical, Quirks: None, Size: 618240kB, Removable

```
The issue still seems to be present with file sizes being truncated when a file is zero-filled. It looks like you have updated to version 2.23 of the SdFat library but in ImageBackingStore you still reference m_fsfile.size(), those calls need to be updated to the new m_fsfile.dataLength() which is what should return the total allocated length on exFat systems. The size() call still behaves as it did in previous releases.

erichelgeson, 2025-06-27:
Testing this change I actually cant reproduce the initial scenario where it incorrectly reports 0k. What system did you run dd on and what was the full command? I'm on NixOS 25.05 (kernel 6.14) `dd if=/dev/zero of=./HD0-500mbtest.hda bs=500M count=1` Both old and patched fw show the right size. While dd exits quickly I can not eject the device till the file system was synced (says data is being written please wait) - I just want to make sure I can reproduce this locally.

brijohn, 2025-06-27:
So this is a Linux Mint 22.1 system, running kernel 6.8.0-59-generic. And while i used dd to create the zero'd image i then used cp to copy it to the SD card instead of dding directly on the card. So you might try doing a cp to copy the image instead of creating it directly with dd on the SD card.

brijohn, 2025-06-29:
Ok so i did a quick test and the original issue seems to work, the file size is reported correctly.

However now any files that have a validDataLength less then the dataLength, are now reporting as Fragmented. This seems to be due to the fact that the ExFatFile::contiguousRange function in SdFat incorrectly sets the endSector based on the validDataLength and the dataLength field. 

```c
bool ExFatFile::contiguousRange(uint32_t* bgnSector, uint32_t* endSector) {
  if (!isContiguous()) {
    return false;
  }
  if (bgnSector) {
    *bgnSector = firstSector();
  }
  if (endSector) {
    *endSector =
        firstSector() + ((m_validLength - 1) >> m_vol->bytesPerSectorShift());
  }
  return true;
}
```
This causes BlueSCSI to decide the file is not contiguous even though it really is. That would not be a big problem in and of itself, from what i can tell all that means is that blueSCSI will do file access via SdFat instead of doing direct sector access of the image file. 

However after discovering this I went and checked the ExFatFile::read and ExFatFile::seek functions and both of those also handle validDataLength wrong as well. read will improperly truncate your read at validDataLength if reading beyond that point, so for a zero filled file it won't actually let you read any data, this probably explains why i still fail to partition the image under dos using afdisk.exe. Seek will return an error if you try to seek past validDataLength. According the microsofts own specs you should be able to seek/read past validDataLength, it just that in that case read should return 0s for any data beyond the validDataLength.

> The ValidDataLength field shall describe how far into the data stream user data has been written. Implementations shall update this field as they write data further out into the data stream. On the storage media, the data between the valid data length and the data length of the data stream is undefined. Implementations shall return zeroes for read operations beyond the valid data length.
>
> If the corresponding File directory entry describes a directory, then the only valid value for this field is equal to the value of the DataLength field. Otherwise, the range of valid values for this field shall be:
> 
>  * At least 0, which means no user data has been written out to the data stream
>
>  * At most DataLength, which means user data has been written out to the entire length of the data stream

So yeah while the file size is being reported correctly I'm not sure I can trust what SdFat is doing with my data currently. I will probably post an issue with upstream about the problems with how they are treating validDataLength and see if that can be fixed. 

One possible solution for the moment would be to update your fork's copy of the ExFatFile::contiguousRange to properly use m_dataLength for endSector calculation, that should get BlueSCSI considering the file to be contiguous and as such bypass the use of SdFat for image access.

This has been a much deeper rabbit hole then I originally imagined. 😃

erichelgeson, 2025-06-29:
Wow, great detective work! And yes a much deeper hole but good some great leads on how to address image fragmentation and correctness on exfat. I've pushed a branch to test your proposed fix for contiguousRange if you'd like to test just update platformio.ini `SdFat=https://github.com/BlueSCSI/SdFat#2.2.3-gpt-testexfatlen`

I agree upstream should really address these changes first and we'll pull them in as there's a lot to consider and greiman likely can give guidance once the issue has been brought to their attention.

brijohn, 2025-06-29:
The patch seems to work ok, I added a bit of logging into BlueSCSI to make sure i was getting the expected values for begin and end and it appears to work as intended and determine the file is in fact contiguous. I can also correctly partition my zero-filled disk image instead of getting an error like before. 

```
Platform: BlueSCSI
FW Version: 2025.06.00-dev Jun 29 2025 15:42:52
Flash chip size: 2048 kB
 
=== SD Card Info ===
SD card detected, exFAT volume size: 119844 MB
SD Name: USD00, MID: 0xAD, OID: 0x4C 0x53

=== Global Config ===
Reading configuration from bluescsi.ini
Using system preset "Generic"

=== Finding images ===
== Opening 'CD2/Fallout 1.1.iso' for ID 2
---- Contiguous image from 0x00252C00 to 0x003809FF sector count 0x0012DE00
---- Configuring as CD-ROM drive
=== Finding images in / ===
== Opening /HD00_512_512MB_DOS622.img for ID:0 LUN:0
---- Contiguous image from 0x00003500 to 0x0010357F sector count 0x00100080
---- Configuring as disk drive
== Opening /HD10_512_512MB_DOS_Games.img for ID:1 LUN:0
---- Contiguous image from 0x00103600 to 0x002035FF sector count 0x00100000
---- Configuring as disk drive
== Opening /HD50_512_512MB_Test2.img for ID:5 LUN:0
---- Contiguous image from 0x00380A00 to 0x004809FF sector count 0x00100000
---- Configuring as disk drive
== Opening /HD40_512_512MB_Test.img for ID:4 LUN:0
---- Contiguous image from 0x00480A00 to 0x005809FF sector count 0x00100000
---- Configuring as disk drive

== Platform supports ROM drive up to 1466 kB
 
=== Configured SCSI Devices ===
ID: 0, BlockSize: 512, Type: Fixed, Quirks: None, Size: 524352kB
ID: 1, BlockSize: 512, Type: Fixed, Quirks: None, Size: 524288kB
ID: 2, BlockSize: 2048, Type: Optical, Quirks: None, Size: 618240kB, Removable
ID: 4, BlockSize: 512, Type: Fixed, Quirks: None, Size: 524288kB
ID: 5, BlockSize: 512, Type: Fixed, Quirks: None, Size: 524288kB
```

This seems to work now as long as the image is contiguous and does not go through the SdFat library for image access. I've filed an issue upstream at https://github.com/greiman/SdFat/issues/527, about the larger issues with how SdFat is handling ExFAT filesystems.

commit 9063091b (2025-06-28): sdfat: Use dataLength() instead of size() to get correct file size no matter the underlying filesystem. Fixes #258