BlueSCSI/BlueSCSI-v2 #272: Recognize OLYMPUS MOS330 as removable (MO) disk drive when in Initiator mode

martinborgman, 2025-06-27:
At present BlueSCSI does not recognize the OLYMPUS MOS330 drive as a removable (MO) disk drive when in initiator mode. This means that BlueSCSI doen't eject the MO disk when finished nor start to read the next drive when a new disk is inserted. It will simply ignore the drive after reading the first disk.

```
Platform: BlueSCSI
FW Version: 2024.12.08-rel Dec  8 2024 22:19:03
I2C Supported
Flash chip size: 2048 kB
SCSI Initiator Mode.  Will scan the bus for drives to image.
 
=== SD Card Info ===
SD card detected, exFAT volume size: 119264 MB
SD Name: USDU1, MID: 0x74, OID: 0x4A 0x60
InitiatorID set to ID 7
 
Initialization complete!
INFO: Pico Voltage: 3.276V.
* No response from SCSI ID 0
* No response from SCSI ID 1
* No response from SCSI ID 2
RequestSense response: 0x70 0x00 0x06 0x00 0x00 0x00 0x00 0x0E 0x00 0x00 0x00 0x00 0x28 0x00 0x00 0x00 0x00 0x00 
Target 3 reports UNIT_ATTENTION, running INQUIRY
SCSI ID 3 capacity 446325 sectors x 512 bytes
SCSI-2: Vendor: OLYMPUS , Product: MOS330          , Version: 1.30
Drive total size is 217 MiB
Unhandled device type: 0x07. Handling it as Direct Access Device.
Imaging filename: HD3_imaged-014.hda.
Preallocating image file
Starting to copy drive data to HD3_imaged-014.hda
SCSI read succeeded, sectors done: 512 / 446325 speed 791 kB/s - 0%
SCSI read succeeded, sectors done: 1024 / 446325 speed 981 kB/s - 0%
.
.
.
SCSI read succeeded, sectors done: 445952 / 446325 speed 1448 kB/s - 99%
SCSI read succeeded, sectors done: 446325 / 446325 speed 1344 kB/s - 100%
Finished imaging drive with id 3
Marking this ID as imaged, wont ask it again.
* No response from SCSI ID 4
* No response from SCSI ID 5
* No response from SCSI ID 6
* No response from SCSI ID 0
* No response from SCSI ID 1
* No response from SCSI ID 2
* No response from SCSI ID 4
* No response from SCSI ID 5
* No response from SCSI ID 6
.
.
.
```

erichelgeson, 2025-06-27:
This will be fixed in the `2025-June-PreRelease` - though currently initiator is commented out to fix something - I'll ping here when it's ready to test.

martinborgman, 2025-06-30:
I haven't tested this extensively but it seems to work.
```
.
.
.
SCSI read succeeded, sectors done: 248826 / 248826 speed 827 kB/s - 100%
Ejecting media on SCSI ID: 3
Finished imaging drive with id 3
Scanning SCSI ID 4
Scanning SCSI ID 5
Scanning SCSI ID 6
Scanning SCSI ID 0
Scanning SCSI ID 1
Scanning SCSI ID 2
Scanning SCSI ID 3
RequestSense response: 0x70 0x00 0x02 0x00 0x00 0x00 0x00 0x0E 0x00 0x00 0x00 0x00 0x3A 0x00 0x00 0x00 0x00 0x00  sense_key 2 asc 0x3A ascq 0x00
RequestSense response: 0x70 0x00 0x02 0x00 0x00 0x00 0x00 0x0E 0x00 0x00 0x00 0x00 0x3A 0x00 0x00 0x00 0x00 0x00  sense_key 2 asc 0x3A ascq 0x00
-- START STOP UNIT on target 3 failed with status 2 and sense_key 0x02 (NOT_READY)
RequestSense response: 0x70 0x00 0x02 0x00 0x00 0x00 0x00 0x0E 0x00 0x00 0x00 0x00 0x3A 0x00 0x00 0x00 0x00 0x00  sense_key 2 asc 0x3A ascq 0x00
RequestSense response: 0x70 0x00 0x02 0x00 0x00 0x00 0x00 0x0E 0x00 0x00 0x00 0x00 0x3A 0x00 0x00 0x00 0x00 0x00  sense_key 2 asc 0x3A ascq 0x00
-- START STOP UNIT on target 3 failed with status 2 and sense_key 0x02 (NOT_READY)
Scanning SCSI ID 4
Scanning SCSI ID 5
Scanning SCSI ID 6
```

If you have the Pico USB connected to your computer, to follow the copy proces, BlueSCSI presents itself as USB storage device by default even when in Initiator mode. This means you first need to eject the BlueSCSI drive before BlueSCSI starts to read any disks.

erichelgeson, 2025-07-12:
Closing as we've cut the release with the code you tested, thanks again.

closed 2025-07-12 by erichelgeson