BlueSCSI/BlueSCSI-v2 #276: Initiator mode no longer works after updating BlueSCSIv2 - 2023.10a to BlueSCSI_Universal_2025-07-08_23b44ab6.uf2

fdanapfel, 2025-07-09:
After updating to the latest release I'm no longer able to image disks using initator mode with a BlueSCSIv2 - 2023.10a Desktop with a Pico 1 W.

What happens is the following:
- when there is no image on the SD card then no log.txt is written on the SD card and the LED on the Pico lights up for about 10 to 15 seconds and then goes off again and nothing else happens
- if a disk image already exists on the SD card then the log.txt is written to the SD card, but it shows some errors trying to access the attached disk:
```
Platform: BlueSCSI (Pico1/W)
FW Version: 2025.07.08-rel Jul  9 2025 02:11:20
Flash chip size: 2048 kB
I2C Supported
 
=== SD Card Info ===
SD card detected, exFAT volume size: 60874 MB
SD Name: SD64G, MID: 0x03, OID: 0x53 0x44
Speed grade set to  reclocking system
Initial Clock set to 133MHz
Reclocking the MCU to 200MHz
Setting the SDIO clock to 40MHz
After reclocking, system reports clock set to 200MHz
======== Reinitializing BlueSCSI after reclock ========
[710ms] 
[710ms] === Global Config ===
[710ms] Reading configuration from bluescsi.ini
[711ms] DBG -- SelectionDelay = 255
[711ms] DBG -- EnableUnitAttention = No
[712ms] DBG -- EnableSCSI2 = Yes
[712ms] DBG -- EnableSelLatch = No
[712ms] DBG -- MapLunsToIDs = No
[712ms] DBG -- EnableParity = Yes
[712ms] 
[712ms] === Finding images ===
[774ms] === Finding images in / ===
[774ms] DBG -- Using block size, 733 from filename: HD20_733MB_imaged.hda
[775ms] == Opening /HD20_733MB_imaged.hda for ID:2 LUN:0
[777ms] DBG ---- Image file is contiguous, SD card sectors 11451904 to 12883663
[777ms] ---- Configuring as disk drive
[777ms] DBG ---- Geometry set from defaults: SectorsPerTrack=63 HeadsPerCylinder=255 total sectors 1000083 (not divisible)
[777ms] DBG ---- Read prefetch enabled: 4608 bytes
[779ms] 
[779ms] == Platform supports ROM drive up to 1465 kB
[779ms] DBG ---- ROM drive image not detected
[779ms]  
[779ms] === Configured SCSI Devices ===
[779ms] ID: 2, BlockSize: 733, Type: Fixed, Quirks: Apple, Size: 715879kB
[851ms] 
[864ms] SCSI Initiator Mode
[1456ms] Clock set to: 200MHz
[1456ms] Initialization complete!
[1459ms] Scanning SCSI ID 1
[1459ms] INFO: Pico Voltage: 3.250V.
[2460ms] DBG -- BUS_BUSY
[2460ms] DBG ---- SELECTION
[2460ms] DBG ------ SELECTING 1 with initiator ID 0
[47456ms] --------------
[47456ms] WATCHDOG TIMEOUT, attempting bus reset
[47456ms] Platform: BlueSCSI
[47456ms] FW Version: 2025.07.08-rel Jul  9 2025 02:11:20
[47456ms] GPIO states: out 0x1EFFFCFC oe 0x0EA08E00 in 0x1EEFFDFF
[47456ms] scsiDev.cdb: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[47456ms] scsiDev.phase: -1
[47456ms] SCSI DMA state: IDLE
[47457ms] Current buffer: 0x00000000/0x00000000, next 0x00000000 bytes
[47457ms] SyncOffset: 0 SyncPeriod 0
[47457ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[47457ms] PIO Data SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[47457ms] PIO Sync SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[47457ms] DMA CH A: ctrl: 0x00000000 count: 0x00000000
[47457ms] DMA CH B: ctrl: 0x00000000 count: 0x00000000
[47457ms] DMA CH C: ctrl: 0x00000000 count: 0x00000000
[47457ms] DMA CH D: ctrl: 0x00000000 count: 0x00000000
[47457ms] GPIO states: 0x1EEFF9FF
[47457ms] STACK 0x2003FDC0:    0x10039C9B 0x1EEFFDFF 0x2001BFA4 0x20016DBF
[47457ms] STACK 0x2003FDD0:    0x10039C9B 0x00000000 0x2003FE40 0x00000000
[47457ms] STACK 0x2003FDE0:    0x20033ACC 0x0000000C 0x02D421C6 0x2003A391
[47457ms] STACK 0x2003FDF0:    0xD0000128 0x00000000 0x00000000 0x20014651
[47457ms] STACK 0x2003FE00:    0x00000000 0x2003A391 0xD0000000 0x00000000
[47457ms] STACK 0x2003FE10:    0x2001F098 0x000009A0 0x00000000 0xFFFFFFF9
[47457ms] STACK 0x2003FE20:    0x00000000 0x00080000 0xD0000000 0xF77FCFF8
[47457ms] STACK 0x2003FE30:    0x2000EBF5 0x100204EB 0x2000EC54 0x61000000
[47457ms] DBG -- BUS_FREE
[57460ms] Target is holding BSY for unexpectedly long, running reset.
[57714ms] DBG startstop: FAILED
[57714ms] DBG read capacity: FAILED
[57714ms] DBG inquiry: FAILED
[57718ms] Scanning SCSI ID 2
[58719ms] DBG -- BUS_BUSY
[58719ms] DBG ---- SELECTION
[58719ms] DBG ------ SELECTING 2 with initiator ID 0
[67622ms] DBG -- BUS_FREE
[67624ms] DBG startstop: FAILED
[67624ms] DBG read capacity: FAILED
[67624ms] DBG inquiry: FAILED
```

If I switch back to https://github.com/BlueSCSI/BlueSCSI-v2/releases/tag/v2024.12.08, keeping the rest of the setup exactly the same, Initiator mode works without issues again (also the log.txt is written to the SD card even if no disk image already exists):
```
Platform: BlueSCSI
FW Version: 2024.12.08-rel Dec  8 2024 22:19:03
I2C Supported
Flash chip size: 2048 kB
SCSI Initiator Mode.  Will scan the bus for drives to image.
 
=== SD Card Info ===
SD card detected, exFAT volume size: 60874 MB
SD Name: SD64G, MID: 0x03, OID: 0x53 0x44
[443ms] InitiatorID set to ID 7
[1592ms]  
[1592ms] Initialization complete!
[1607ms] INFO: Pico Voltage: 3.254V.
[2860ms] * No response from SCSI ID 0
[3874ms] RequestSense response: 0x70 0x00 0x06 0x00 0x00 0x00 0x00 0x18 0x00 0x00 0x00 0x00 0x29 0x00 0x00 0x00 0x00 0x00 
[3874ms] Target 1 reports UNIT_ATTENTION, running INQUIRY
[10100ms] SCSI ID 1 capacity 8925000 sectors x 512 bytes
[10100ms] SCSI-2: Vendor: IBM     , Product: DDRS-34560      , Version: S97B
[10101ms] Drive total size is 4357 MiB
[10158ms] Imaging filename: HD1_imaged-001.hda.
[10161ms] Preallocating image file
[10204ms] Starting to copy drive data to HD1_imaged-001.hda
[10359ms] SCSI read succeeded, sectors done: 512 / 8925000 speed 1759 kB/s - 0%
[10511ms] SCSI read succeeded, sectors done: 1024 / 8925000 speed 1783 kB/s - 0%
[10663ms] SCSI read succeeded, sectors done: 1536 / 8925000 speed 1795 kB/s - 0%
[10810ms] SCSI read succeeded, sectors done: 2048 / 8925000 speed 1833 kB/s - 0%
...
[2594104ms] SCSI read succeeded, sectors done: 8924160 / 8925000 speed 1927 kB/s - 99%
[2594247ms] SCSI read succeeded, sectors done: 8924672 / 8925000 speed 1885 kB/s - 99%
[2594341ms] SCSI read succeeded, sectors done: 8925000 / 8925000 speed 1865 kB/s - 100%
[2594471ms] Finished imaging drive with id 1
[2594472ms] Marking this ID as imaged, wont ask it again.
[2595731ms] * No response from SCSI ID 2
```

The bluescsi.ini looks like the following:
```
[SCSI]
InitiatorMode=1
EnableUSBMassStorage=false
Debug=1
```

erichelgeson, 2025-07-10:
Thanks for the report! Could you test this build and let me know if it works? https://github.com/BlueSCSI/BlueSCSI-v2/actions/runs/16204056404 - Click on the "BlueSCSI binaries" to download, flash the uf2. Rerun test.

erichelgeson, 2025-07-11:
[image]

Found the issue. I'll push this and another bugfix tomorrow - getting too late here! Thanks again for the report.

erichelgeson, 2025-07-11:
https://github.com/BlueSCSI/BlueSCSI-v2/releases/tag/v2025.07.11

closed 2025-07-11 by erichelgeson

fdanapfel, 2025-07-12:
@erichelgeson Thanks for the fix. Image mode works again with [v2025.07.11](https://github.com/BlueSCSI/BlueSCSI-v2/releases/tag/v2025.07.11)

However I noticed the following issues:
- the green LED on the PicoW now stays permanently on and does not change when a disk is imaged (the blue activity LED on the BlueSCSIv2 board only blinks very faintly when Initiator mode is enabled; this makes it difficult to see if a disk is being imaged if the disk itself has no activity LED)
- if there is no image on the SD card, no log.txt is written on the SD card when Initiator mode is enabled. Even if "Debug = 1" is set in the bluescsi.ini (could this be related to https://github.com/BlueSCSI/BlueSCSI-v2/wiki/RAW-Mode ? If yes, it might make sense to disable RAW mode when Initiator mode is enabled)

erichelgeson, 2025-07-12:
#1 - yes, there is an issue with the Pico-W LED and any SCSI activity - and it actually slows down everything quite a bit. The Pico-W uses the RM2 WiFi module to toggle the LED which is extremely slow. I recommend connecting to a computer console to watch the logs live. We will have a i2c screen/buttons to interact with initiator in the future as well as even just "watching the led" is not really enough to know whats going on.

#2 ah! good catch. Yes it's probably falling into raw mode. I'll add a skip to that for next release. Work around is to just have a dummy HDx.hda on the root for now.

fdanapfel, 2025-07-19:
@erichelgeson 
> Correct, the Pico(non-W) LED is just driven by GPIO like a sane solution would :D. Note you cal always attach a LED to the LED header on the BlueSCSI to get LED feedback, but again, console out is much better - I'm actually adding in a lot of commands to control initiator mode from your computer while it's running.

Thanks again for the clarification.

Maybe you could add this info to https://github.com/BlueSCSI/BlueSCSI-v2/wiki/Initiator-Mode

Or at leasr mention that when using a Pico-W the LED won‘t work in Initiator mode.

erichelgeson, 2025-08-03:
Whoops! Thanks for the reminder - It'll be in the next one now for sure- just pushed it to main-new - https://github.com/BlueSCSI/BlueSCSI-v2/commit/e45dcb395d7e617a60f69db59b31eac912f5ecea

fdanapfel, 2025-08-05:
@erichelgeson 
> Whoops! Thanks for the reminder - It'll be in the next one now for sure- just pushed it to main-new - [e45dcb3](https://github.com/BlueSCSI/BlueSCSI-v2/commit/e45dcb395d7e617a60f69db59b31eac912f5ecea)

I just tested the firmware with this commit from https://github.com/BlueSCSI/BlueSCSI-v2/actions/runs/16710303034 but now it looks like there is no log.txt written for both target mode and initiator mode if there is no disk image file on the SD card.

Also having an empty HD0_dummy.hda file on the SD card doesn't seem to work, only when a real disk image on the SD card the BlueSCSIv2 creates the log.txt when in initiator mode.

commit 7874a6ae (2025-07-11): Fix for initiator initialization. Fixes #276
Set initiator file handling to previous default.
Other minor cleanup.